bigtreetech/BIGTREETECH-TouchScreenFirmware #870: Unknown command setting LED colour.

sam01508, 2020-07-26:
When trying to change the colour of the led for the rotary encoder you get an unknown error M150 RO UO B255 P255. Any menu item in the led section gives you this error. You can activate from the machine menu? and change the colour but not in the actual led colour section.
You also can't toggle between Marlin mode and Bigtreetech mode using the rotary encoder.

Led should change colour or be turned on and off.

Error as above.

TFT35 V3.0 E3
Main board is SKR mini E3 V2
Latest Bugfix version of Marlin on the main board, Screen TFT35 precompiled firmware bin, also acted the same with firmware that came with the screen.

oldman4U, 2020-07-26:
You likely use the wrong control. Can you mAke a Picture  which shows the screen where you try to change the color.

oldman4U, 2020-07-26:
It is not possible to use encoder to change the mode but use the touchscreen to do so.

sam01508, 2020-07-26:
Thanks for your input, I think I'm looking at the RGB lighting control I'll try and put a picture up tomorrow just for confirmation. Regarding the rotary encoder, I had seen it used to toggle between the two modes, and then when mine didn't I thought it was an issue. Thanks again.

oldman4U, 2020-07-26:
Send the picture.

As far as i know the selection using the encoder has been removed because
using the encoder for selecting also changed the feedrate in the
background.

Tygrys-1, 2020-07-27:
The "LED color" item in "Machine" menu is used to change the color of Neopixels attached to the printer.  You have to have leds attached and compile Marlin with the coresponding #define enabled, otherwise the M150 command is not recognized. To change the color of the knob backlight use the "Features" menu.

Encoder is not working on the mode change menu anymore, see #708.

sam01508, 2020-07-27:
Great thank you that's the thinking I had once I'd had the chance to study it better. Thanks to everyone who took the time to comment I'll close this now.

closed 2020-07-27 by sam01508