bigtreetech/BIGTREETECH-TouchScreenFirmware #871: Text gets scrambled when preheating

Whiskaya, 2020-07-27:
### I started this as an SKR Mini issue, but I believe it belongs here now, as pretty much everything about the SKR Mini has been discarded so far.

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/335

**Description**

Screen scrambles text and shows unintelligible symbols that change fast when starting a print.
Sometimes this happens when starting in TFT mode and then switching to Marlin mode, without starting a print.
Sometimes text scrambles lightly in Marlin mode. Moving the knob may sometimes fix it.
(VIDEOS AT THE END)

**Steps to reproduce**

Start printer normally (in Marlin mode).
Start preheat.
Start print.
Expected behavior
Normal printing.

Actual behavior
Printer keeps working as normal.
Entering menus (if you remember the moves on the knob) is still possible.
Screen is unreadable.

**Actions taken**

Flashed stock (stable) firmware on main board and TFT board.
Changed ribbon cable.
Tried different SD cards, in all formats (FAT16 and FAT32) and sizes (micro w/ adapter and regular), from board and TFT mount.
Connected stock display.

**Hardware**

Ender 5
Stock PSU
SKR Mini E3 V2.0
TFT35 E3 V3.0

radek8, 2020-07-27:
Switch off the PWM control in the Marlin setting, on the bed heating.
If this does not help, it will be a defect of the HW display. Complain to the seller.

Whiskaya, 2020-08-01:
I think it has nothing to do with the PWM, as I just tried printing from USB, and the screen shows no scrambling.
Then again, 95% sure this is firmware.
[image]

radek8, 2020-08-02:
212/5000
Everyone uses this FW, but no one has your problem.
The problem will be in HW.
The bed has a large stream.
If you have a weak power supply, PWM could signal CPU interference
It's just one of the options to check

radek8, 2020-08-02:
The TFT display may also be defective

jphansendk, 2020-08-10:
Any update on this topic? I still have this problem as well.

oldman4U, 2020-09-11:
Hi.

There has been a topic in Marlin which could be related to this issue, where high frequency PWM causes issues.

Do you use such setting in your firmware?

Whiskaya, 2020-09-29:
Tried changing FAST and SOFT PWM... nothing.

Also updated to a Mean Well PSU... nothing.

thebiggestgit, 2020-11-29:
Interesting, as I've tried this board with my Ender 5 Pro and stock Creality v1.15 8-bit board (before I've replaced that). The TFT 35 display becomes scrambled during printing after a few minutes each time (apart from line at top , note this is when using a USB printer connection with a RaspberryPi 4 and Octoprint - I'll need to test to see if also happens via SD card.  This was both on the stock TFT firmware device came with TFT35_E3_V3.0.26.x Jun 15 2020 and latest Nov 27 2929 versions. Not however really an issue as such for me as I still need to install the SKR Mini E3 V2.0 board to go with this new display, and would expect to use in the touch screen mode, but will re-test once that's upgraded and report back if any issues with the ender 5 pro still. I tried all of the serial port settings in the touch screen config and made no difference - not even sure that does anything in Marlin mode?

oldman4U, 2020-11-29:
Try using well shielded cables instead of the one which are provided with the TFT/Mainboard. This particular ticket is related to preheating and not while printing, but the reason for it is most likely interference in both cases.

thebiggestgit, 2020-11-30:
Thanks I found for me this was related to having the Raspberry Pi connected to the ender 5 pro creality board via USB and causing interference when using the printer.

oldman4U, 2020-11-30:
Perfect.

Happy printing

MarkMan0, 2021-01-29:
Related issue, using the master branch, text gets scrambled similar to OP. Board is SKR PRO V1.1, with fairly long cables. The issue is **not** present using x.x.26 version. Scrambling occurs when heating - I assume some kind of interference with the heater PWM. Printer is a custom made and fairly big - hence the need for long cables.
Again, no scrambling with version x.x.26, and also with reprap full graphics lcd. Only with the master branch.
TFT 35 E3 V3.0

softwarecrash, 2021-02-06:
> Related issue, using the master branch, text gets scrambled similar to OP. Board is SKR PRO V1.1, with fairly long cables. The issue is **not** present using x.x.26 version. Scrambling occurs when [...]

Did you mean the x.x.26.1 Release from 9 month ago?
Have you testet the actual Release?
I have the Same Problem, geht smashed Display when the Big z Motor start driving

MarkMan0, 2021-02-08:
I know commit `3e75031 Release xx.26 (#566)` is working, haven't tested `a60960c Release   xx.26.1 (#653)`. The display can with xx.26 firmware, so I knew that was good, I've been using it for some time, before trying to upgrade.
I've tried to bisect, but unfortunately I couldn't pinpoint the commit that caused the issue. I know commit `f193337   Update language_ru.ini (#1222)` is **bad** for sure, and I assume everything after it too. Right now, my goal is to find the commit which introduced this behavior. But as you can imagine, this takes a long time, because the issue doesn't always come out, so I can't say with 100%, if the given commit is good.  The latest good commit I tested is `a599575 Update language_cz.h (#1002)`. 
Could you tell me which version you tested, and had the issue with?

digant73, 2021-04-09:
@Whiskaya that kind of problem is HW and typically are due to EMI. Some printers have better shild than other. The power supply produces EMI. I had that kind of problem and I fixed it using shielded cables and ferrite bead

oldman4U, 2021-04-13:
@Msq001

Like digant73 explained finally. This is not a firmware issue but related to hardware.

Please close this ticket

Thank you

closed 2021-04-13 by bigtreetech