bigtreetech/BIGTREETECH-TouchScreenFirmware #881: Scrolling with rotary encoder in Terminal window

hackra76, 2020-07-29:
Hi,
just a suggestion. Is it possible to program the Terminal window in touch mode so that the rotary encoder scrolls through the displayed content in the terminal window ?

For example, if I issue a M503 it lists all the configured items, but the list is really long and so you can see only few last lines. I would suggest to use the rotary encoder (knob) to scroll up to see the older messages...

Thank you

radek8, 2020-07-29:
Use the encoder only as a last resort.
It has side effects

radek8, 2020-07-29:
In touch mode, marlin responds to encoder rotation and changes the print speed.
You would have to disable this feature in Marlin, or ensure that marlin is not on the home screen.
The resulting effect is not responsible for the problems.

Tygrys-1, 2020-07-29:
@radek8 If it is not on the home screen one can do even more harm... There should be some switch in Marlin to "ignore the encoder untill further notice". I.e. a very long press or something like that. It will unlock the potencial of encoder in touch mode.

The encoder is hardwired to the Marlin, so each move or press acts on the Marlin menus regardless of which interface is displayed. So using the encoder in touch mode is kind of a lottery under the hood. I know, that it will be usefull for setting values or navigation in touch mode, but in the current hardware implementation it is a no go.

charakaSamawry, 2020-08-20:
Ooooooh. That would explain why my speed always changes 'on its own'
There should be an instruction somewhere that warns about using the encoder to set values. Had noticed whenever i set temps or fan speeds or z offset etc, the speed had changed the same amount, or there abouts. 

I had been using the encoder wheel because tapping a touchscreen more than 10 times is less effective than spinning the wheel.. 
I also sometimes try to scroll the menus using the encoder wheel lol thought that would be cool for when im already using the encoder wheel but not at all worth the trouble.

oldman4U, 2020-09-03:
@hackra76 

This topic is covered in the pinned tickets at the first page of this repo and also several other opened and closed tickets have the same content.

Please close this ticket.

Thank you

oldman4U, 2020-09-11:
@bigtreetech 

Please close this ticket, because the topic is covered in the Readme.

Thank you

closed 2020-09-11 by hackra76

oldman4U, 2020-09-11:
👍🏻