bigtreetech/BIGTREETECH-TouchScreenFirmware #883: F0 value set in Marlin mode is not updated in Touch mode.

Paul-Anon, 2020-07-29:
The part fan (F0) value set in Marlin mode does not get sent back to Touch mode when switching.
(However, F0 value set in Touch mode does get successfully transferred to Marlin mode.)

radek8, 2020-07-29:
You used a precompiled FW that does not contain all the fixes.
Compile your own FW from source files.

Paul-Anon, 2020-07-30:
No, you're wrong. I have never used a precompiled .bin file!
The 'Master' was updated only 16 hours ago so I guessed that was the latest.
As this one (I've no idea what the version is) gave the problem I tried compiling the only other I found - release-xx.26.1 (which hasn't been updated for 3 months even though it appears a later version). This displayed the exact same problem and introduced another - it ignores the config.ini file completely.

I am quite happy to accept that I may have made a mistake - I certainly cannot make head nor tail of Github!

I would be grateful, if I have selected the wrong source files, if you would tell me how to find the correct ones.

As of this moment the original fan problem still exists and I have been unable to get the config.ini file to be actioned.

Any further ideas anyone?

radek8, 2020-07-30:
What display model do you have?

Paul-Anon, 2020-07-30:
I have the TFT35 V3.0. I have now reverted to the 'Master' release and the config.ini issue is resolved. It is doing exactly what it should. The fan F0 is still not indicating. However, I have been testing today and found that, when in Touch mode, it will respond to Marlin commands *when it is printing* but does not when it is idle.

As a quick check, try the following: connect to your printer (in Marlin mode) via USB and send, from Pronterface or whatever, M106 S255

If all is well, F0 will turn on at full speed and Marlin will display 100% together with animated fan icon.
Switch off fan (M107), switch to Touch mode and send M106 S255 etc as above.
In my case, fan starts as expected but F0 still displays 0%. It should, of course, display 100%

Clearly there is a problem.

radek8, 2020-07-30:
You are using old FW.
Use this FW
[BIGTREE_TFT35_V3.0.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5003132/BIGTREE_TFT35_V3.0.26.x.zip)

Paul-Anon, 2020-07-30:
No this does not work either. I was already using BIGTREE_TFT35_V3.0.26.x from the 
master but I did try both as although they are designated with the same version 
number they are of different sizes. (Version control issue?)
Also, if you add custom menu entries in config.ini:

custom_label_8:Fan Test (F0) Full On
custom_gcode_8:M106 S255\n
custom_label_9:Fan Test (F0) Off
custom_gcode_9:M107\n

you will find that, while the fan responds as it should, the F0 % indicator does not change.
Are you able to replicate?

radek8, 2020-07-30:
It works
the fan status indicates OK
The FW I sent must work
In the ini file, try to choose
serial_always_on: 1

Paul-Anon, 2020-08-01:
I have spent hours struggling with this.
custom_label test is now working - most of the time but is still a little inconsistent.
The original issue still remains and has never worked for me. I have even gone back and checked and recompiled Marlin.
Tested serial_always_on but it didn't help either.
Also, I now find that when I update config.ini I always get invalid keyword for pl_recovery_en:0 message. I didn't see this before.
I am going to fit an SKR 1.4 Turbo and rebuild from the ground up tomorrow so hopefully that will fix the problem.
I'll let you know the outcome.

radek8, 2020-08-01:
> Also, I now find that when I update config.ini I always get invalid keyword for pl_recovery_en:0 message. I didn't see this before.

If you are retrieving this message, you are using the current FW.
@bigtreetech can you please compile the current bin files?
Existing ones have a bug and a lot of users don't bother to compile a new bin file

Paul-Anon, 2020-08-02:
'If you are retrieving this message, you are not using the current FW.' ??

You informed me earlier that I didn't have the latest firmware and consequently sent me 'the latest' (? - I've no idea what version it is as they are all annotated the same.) It was this firmware that eventually produced the  pl_recovery_en:0 message. Why would you send me firmware and then tell me I am not using current firmware?? You said 'The FW I sent must work'? It didn't.
Anyway, I compiled again and the problem remains. I suspect now that the firmware has a number of other bugs that come and go at random. The custom entries that I set to test the original fan issue have inexplicably stopped working now and there are odd freezes too (not to mention a yellow pixel that appears occasionally in the extreme top right corner and blinks).

I am bench testing the new 1.4 Turbo board together with a second new TFT35. I have configured and compiled Marlin 2.0.6. and compiled the TFT35 firmware for testing and comparison with the original Marlin 2.5.4 + TFT35 setup in the printer.
Neither work properly (even down to the odd yellow blinking pixel).The original fan bug persists on both.

I can only assume that, as you are unable to replicate the bug, I have been unable to explain it clearly enough - my fault not yours. That is the only explanation I can think of as you have never had it not work and I, with two setups, have never seen it work!

There is little point in asking BTT to compile the bin files as all my issues are from the source - I don't use precompiled bin files as I explained to you before!

I have to wonder if BTT monitor issues posted here. Any input from BTT would be welcome.

The problem I have now is not so much the fan issue but whether I can have any confidence in what appears to be a bug ridden and very flaky, inconsistent display.

radek8, 2020-08-02:
I assume that the yellow square indicates communication on the serial line.
I sent you a bin file that should address the fan problem. But I can't know if you really used it.
The bottle did not have to go correctly.

There is a new bug in the current issue, see
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/898
there is also a manual for a temporary repair

bigtreetech, 2020-08-03:
This seems to be because Marlin does not provide an interface to query the fan speed, so the touch screen cannot update the real-time speed of the fan setted by other ways(such as: lcd12864, usb or octoprint, and soon...).
We need to improve both TouchScreen and  Marlin‘s program to solve this problem

adamhf, 2020-08-03:
@Paul-Anon For what it's worth, I was having the exact same issue as you.  I was building the TFT firmware from master and using the 2.0.x bugfix branch of Marlin without success.  I did some digging in the code and discovered REPORT_FAN_CHANGE in the [config_adv file](https://github.com/MarlinFirmware/Marlin/blob/5e3411029d594ade5b4a496ac43864b9fa52fbdc/Marlin/Configuration_adv.h#L3114). After enabling this and recompiling I can now see fan speed updates on the TFT in touch screen mode.

Paul-Anon, 2020-08-05:
@adamhf Apologies for late response.
Thank you so much! I've been though the configuration so many times I see it in my sleep but somehow I failed to see that!!

Well, things are now different:
fan temp is working! It appears to be doing everything it should (but it is early days). 
However, some new problems have surfaced although, at this point, I don't know if they are related to this change (I can't imagine why they should be).

1. Occasionally the hotend and bed icons are not updated at end of print even though they are both switched off and temperatures are shown to be falling in pronterface.

2. Most serious: Extruder doesn't always activate so have to abort print and reboot. Mostly happens when printer first switched on. Did happen in the middle of a print on one occasion. I'm sure I've seen this reported as an SKR 1.4 and TMC2088-UART issue.

3. Custom actions don't work at all although they were fine before. (not related to this change).

I have many issues but, thanks to you, the fan problem seems to be solved.

I'm going to try switching Marlin 2.0.6 back to 2.0.5.4 (with your suggestion) and also recompile the latest TFT35 code and see how things go.

guruathwal, 2020-08-11:
The latest marlin release automatically sends fan speed change to all serial ports connected. 
See: https://github.com/MarlinFirmware/Marlin/releases
Updating to latest Marlin release version should resolve this.

sam01508, 2020-08-14:
This issue is still there, well at least for me anyway. I've got the latest marlin firmware 2.0.6 on the skr mini E3 V2 board and the latest (downloaded yesterday) TFT35 E3 V3 firmware. The previous version when compiled in vs code fixed it. However with this updated version from yesterday, it has now returned. I have also compiled this through vs code and it has not fixed it. Are we any closer to a fix for this.

Paul-Anon, 2020-08-15:
@sam01508 you are perfectly correct! The update yesterday has broken it again. I have spent hours getting back to the earlier version. I now have so many 'versions' of this bl**dy awful firmware and struggle to know which is which. It is completely out of control with every man and his dog hacking at it. Quite why BTT allowed things to get to this state is beyond me. This must surely damage their product and reputation. There are plenty of other bugs too but, at this point, I really can't be bothered to officially raise them else 'they' will break something else trying to fix them! I will live with the layer indicator not resetting to zero for example!!

Marlin is not the problem since 2.0.6 introduction of REPORT_FAN_CHANGE. (It could never have worked before this despite some suggesting that it worked fine! BTT confirmed this.) Hopefully you have your previous version and can reload that. Failing that, you will just have to hope that it will get fixed soon. If it does get corrected, be prepared for something else to get broken!

sam01508, 2020-08-15:
@Paul-Anon, It does seem like it could be very damaging to their reputation I agree. For the most part it's good work way beyond my capability. 
I like something to work as it should. I expect the odd bug, but not fix one have another as you put it. Now as I said it comes up with failure to enable bed levelling which is new, ok the error message and it says ABL active go figure. But each time you turn it on its off with the error. I might have to put this through as a bug.

Paul-Anon, 2020-08-15:
@sam01508, hang on - when you install new firmware you lose your bed leveling data don't you? Test it - issue an M503 from terminal and confirm your data is still there. I suspect it will have gone which could explain your error. Try doing another 'auto calibration'. If all is well I would follow that with a 'bed leveling' and finally 'store settings'. I entirely agree with you - I expect it to work and I also except there will be bugs but...

sam01508, 2020-08-15:
> @sam01508, hang on - when you install new firmware you lose your bed leveling data don't you? Test it - issue an M503 from terminal and confirm your data is still there. I suspect it will have gone [...]

I did those steps last night and it still came up with the error. Anyway did it again this morning and it has fixed that.
Just the fan now! I do have my previous configuration so may result back to that in the meantime. Thank you for your help.

sam01508, 2020-08-16:
@guruathwal just an FYI really, the fan speed responds when changed in the menu as half/full + - etc its when printing the fan speed stays at 0. I'm sure you are aware that this is the issue but I have read that it doesn't change when using the menu to access it. Hope that makes sense. Kind regards

Msq001, 2020-08-18:
> @sam01508 you are perfectly correct! The update yesterday has broken it again. I have spent hours getting back to the earlier version. I now have so many 'versions' of this bl**dy awful firmware and [...]

Sorry, #969 breaks the normal display of fan， it should be fixed in #989

Paul-Anon, 2020-08-19:
Thank you, the fan issue appears to be fixed!!!

I only have two further issues at this time:

1. The layer indicator is not reset to 0.00mm at the beginning of a print. It continues to increment from the value of previous print.  (Also, on one occasion, I had it display -20.00mm on an initial print which may suggest that it is not being initialised correctly at switch on.) The layer height is only relevant and useful while a print is in progress. Could it not be reset to zero once a print is completed or cancelled?

2. Both hotend and bed remain powered if a print is cancelled or aborted. They should both be reset to zero in these cases. This is potentially dangerous if it were to go unnoticed.

sam01508, 2020-08-20:
> Thank you, the fan issue appears to be fixed!!!
>
> I only have two further issues at this time:
>
> 1. The layer indicator is not reset to 0.00mm at the beginning of a print. It continues to increment [...]

Hi Paul-Anon, how was the fan issue resolved? I too have noticed the hot end and the bed not cooling down after stopping a print. It would normally shut down heating and move to the side away from the print wouldn't it?

radek8, 2020-08-20:
> Thank you, the fan issue appears to be fixed!!!
>
> I only have two further issues at this time:
>
> 1. The layer indicator is not reset to 0.00mm at the beginning of a print. It continues to increment [...]

But this is not a FW display problem.
The problem is with your slicer configuration.
1.
If you put in the start code G28 executes, a zero alignment will be performed.
2.
If you give the end code the command to turn off the heating and extend the bed
your problem will be solved.

If you do not want to put it in the slicer, use the start code and end code in the display configuration.

Paul-Anon, 2020-08-20:
sam01508
I am confused by the fan issue. Once I implemented REPORT_FAN_CHANGE in Marlin 2.0.6 the problem was solved for me. I was not aware it had been broken again in #969. I guess I had skipped that version. I have no idea how it was resolved but, at least for now, it appears to work. I have tried to break it via Marlin mode and pronterface but I cannot. Excellent!

The issue I mentioned above re hotend/bed remaining powered is an error on my part - mea culpa!
The instruction to switch these off is at the end of the print file gcode. In my case, Simplify3D includes:

M104 S0;     ;extruder heater off
M140 S0;     ;heated bed off
G28;             ;home
M84;            ;steppers off
M107;          ;part fan off

As you might expect, if a print is cancelled this is not executed and the hotend and bed remain powered.
BTT manage this (in the config.ini file) with:

#### Default Cancel Gcode Status
# Options: [enable: 1, disable: 0]
cancel_gcode_enabled:0

#### Cancel G-code - run this G-code after cancelling print
# maximum length 50 characters
cancel_gcode:M104 S0\nM140 S0\nG28 XY R10\n

You can of course tailor 'cancel_gcode' to suit your needs.
This works just fine but note it is disabled by default. However, it can be toggled on/off in the interface:

Menu - Settings - Feature - Cancel Gcode

Apologies to BTT!
I like this solution as it can be left on for a subsequent print if required. Again, I hold my hand up on this one. I should pay more attention. Sorry BTT.

For me, at the moment, it is just the 'Layer' reset/initialise issue.

Paul-Anon, 2020-08-20:
@sam01508 
FYI the part cooling fan, if on, remains running on cancellation so I added M107 to switch it off.

cancel_gcode:M104 S0\nM140 S0\nM107\nG28\n

I find the above works well for me.

Paul-Anon, 2020-08-23:
@sam01580

I have managed to modify the code to start every print at '0.00mm'.
It works 100% of the time for me.
I have included my modified file just in case you would like to try it for yourself.
Note that I have also changed the label title from 'Layer' to 'Height' as I feel that to be more appropriate. It is easily changed so not a problem.
If you do a 'Find' for '[aec]' you can see the changes I have made. I'm not familiar with 'C' so it is likely I have broken every rule in the book but at least I've had a go and, as far as I can tell, it works flawlessly. (Hopefully no unforeseen consequences like memory leaks etc!).

[PrintingMenu.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5115016/PrintingMenu.zip)

sam01508, 2020-08-23:
@Paul-Anon That’s great news to here you have this working. I’m no expert by any means but I’ll take a look and let you know the results should I be able to set it up! Thank you again for taking the trouble to work on this.

Paul-Anon, 2020-08-23:
@sam01580
Sorry, there is still a problem. Don't waste your time compiling this one. I've learnt a lesson here - don't continually use the same test file!!! I'm sure I will crack this sooner or later. I ran seven prints without problem - then tried a different file and it failed. I'm going to keep trying.

Paul-Anon, 2020-08-25:
@sam01508 
Ok, it is fixed now and after two days of testing I cannot break it. Every single print now begins at 0.00mm. I'm happy but will continue to test for a while. Attached is a TFT35V3.0 build (I assume that is what you have) if you would care to give it a try. You should run your config.ini with it for baud rate etc. Usual disclaimer of course!

[BIGTREE_TFT35_V3.0.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5123740/BIGTREE_TFT35_V3.0.26.x.zip)

sam01508, 2020-08-25:
@Paul-Anon I'll give it a try when I get some time to have a look. If I can work out what to do! Thanks for the time you've put in on this.

closed 2020-08-27 by Paul-Anon

Paul-Anon, 2020-08-27:
Fan has been corrected and has been fine since Marlin 2.0.6 introduced REPORT_FAN_CHANGE.
I have fixed the 'Layer' not displaying '0.00mm' at beginning of print. I have been testing for extensively without issue.

hapklaar, 2021-04-16:
> As far as I am aware, REPORT_FAN_CHANGE was introduced in Marlin 2.0.6 master (Thanks to @adamhf for pointing this out). (I don't generally use 'bug fix' versions.) TFT screen (fan issue) would not have worked for any earlier version of Marlin. As I mentioned earlier, by default the line is commented out so would need to be enabled prior to compilation for it to work. Once done the fan issue is resolved.

I have REPORT_FAN_CHANGE enabled in Marlin config, and fan speed updates are also being shown in Marlin mode, but not in Touch mode, at least not on the status screen while printing. This is always on 100% (in my case) even though I hear the fan speed changing.

Am I correct in thinking this should be working with a binary compiled from current code?

PR #977 (2020-08-18): move tuning menu to machine setting and simplify eeprom buttons, Fix fan speed display issue
- Move the tuning menu to machine settings
- Simplify EEPROM buttons.
- Fix fan speed display issue (#883)

Fixes #883