bigtreetech/BIGTREETECH-TouchScreenFirmware #884: [Q] (Your Question?) I have a problem. firmware is not compiled

OrangePrilip, 2020-07-29:
Skr 1.3 + tft 3.5_v3.0

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/STM32F103VC.html
PLATFORM: ST STM32 6.1.0 > STM32F103VC (48k RAM. 256k Flash)
HARDWARE: STM32F103VCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-stm32cube 2.0.181130
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Warning! Cannot find a linker script for the required board! Firmware will be linked with a default linker script!
InvalidJSONFile: Could not load broken JSON: C:\Users\Tom\.platformio\packages\framework-stm32cube\platformio\variants_remap.json:
  File "C:\Users\Tom\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Tom\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Tom\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Tom\.platformio\platforms\ststm32\builder\main.py", line 98:
    target_elf = env.BuildProgram()
  File "C:\Users\Tom\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Tom\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\Tom\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Tom\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 118:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\Tom\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Tom\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 330:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "C:\Users\Tom\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 660:
    return method(*args, **kw)
  File "C:\Users\Tom\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Tom\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Tom\.platformio\platforms\ststm32\builder\frameworks\stm32cube.py", line 233:
    join(FRAMEWORK_DIR, "platformio", "variants_remap.json"))
  File "C:\Users\Tom\.platformio\penv\lib\site-packages\platformio\fs.py", line 56:
    raise exception.InvalidJSONFile(file_path)
=========================================================== [FAILED] Took 1.91 seconds ===========================================================

oldman4U, 2020-08-04:
Hi. Looks like a configuration or installation issue to me. Are you able to compile other firmware successfully?

oldman4U, 2020-09-03:
@OrangePrilip 

Close this unused ticket now or reply.

oldman4U, 2020-09-11:
@bigtreetech 

Please close this ticket.

Thank you

closed 2020-11-09 by OrangePrilip