bigtreetech/BIGTREETECH-TouchScreenFirmware #885: [FR] IDEA: Iconpack 'The Round Miracle' - Current Update for SW v26.x

Acenotass, 2020-07-29:
Added 3 new icons (global_chamber.bmp, chamber.bmp, point_5.bmp) in the current update 26.x

[image]

[TFT24 Iconpack 'The Round Miracle' by Acenotass for v26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4997380/TFT24.Iconpack.The.Round.Miracle.by.Acenotass.for.v26.x.zip)
[TFT28 Iconpack 'The Round Miracle' by Acenotass for v26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4997381/TFT28.Iconpack.The.Round.Miracle.by.Acenotass.for.v26.x.zip)
[TFT35 Iconpack 'The Round Miracle' by Acenotass for v26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4997382/TFT35.Iconpack.The.Round.Miracle.by.Acenotass.for.v26.x.zip)
[TFT70 Iconpack 'The Round Miracle' by Acenotass for v26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4997383/TFT70.Iconpack.The.Round.Miracle.by.Acenotass.for.v26.x.zip)

Old-Trekee, 2020-07-29:
Any chance of adding for the TFT43?

DrumClock, 2020-07-30:
Can you please add also for FTF 50 V3.0?

labotecno, 2020-07-30:
great update !

DrumClock, 2020-08-01:
Thanks, today a few new icons have been added for TFT 50.
Can you please add them?

radek8, 2020-08-01:
@bigtreetech, can you put this icon theme on the main download page?
I think they are the best and used :-)

Acenotass, 2020-08-02:
Done!

Old-Trekee, 2020-08-02:
Thanks much Acenotass.

Mistigris18, 2020-08-02:
Super thèmes, merci 👍

DrumClock, 2020-08-02:
Hi, @Acenotass 
please is it possible to get source images of icons for TFT 50 ?
Thanks

Acenotass, 2020-08-02:
Hi! @DrumClock Source images for TFT50 is the images of TFT35

DrumClock, 2020-08-03:
Thanks @Acenotass 
I meant source images. If I wanted to create my own icon. 
For example in Adobe Illustrator or Corel Draw format ... etc

Acenotass, 2020-08-03:
I am using GIMP, as indicated in the readme.md file

charakaSamawry, 2020-08-07:
These icons werent replaced when i used your iconpack acenotass. Not sure if your just yet to update them or missed them. (Or they are no longer used or are renamed?)

Acenotass, 2020-08-07:
@charakaSamawry https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/935

Acenotass, 2020-08-07:
Obsolete

closed 2020-08-07 by Acenotass

charakaSamawry, 2020-08-07:
@Acenotass Many Thanks!