bigtreetech/BIGTREETECH-TouchScreenFirmware #886: [BUG] (short description)

NicoletoAerials, 2020-07-29:
### Description

### Steps to reproduce
1. Make sure to take note of the initial speed value
2. Go to movement, move, and move any axis via the knob with a fast rotation.
4. Go back to the main menu and check the speed value
5. You will notice that the speed value has incremented o decremented according to the movement made in step 2

**Expected behavior**
Only the values presented on the screen should change via the knob.

**Actual behavior**
By moving the knob in the move menu, the print speed value also gets modified 

### Hardware Variant
[Video.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4997554/Video.zip)

TFT 3.5 V3.0 E3
SKR Mini E3 V1.2
Latest master (29/07/2020)

radek8, 2020-07-29:
look at #881

gordo3di, 2020-08-03:
I have this bug too and it comes and goes... Sometimes it affects the speed and sometimes it doesn't when babystepping. Still trying to pinpoint the series of events that make it happen.

radek8, 2020-08-03:
It depends on what the Marlin screen is showing.
If there is a default screen - it affects the print speed
Menu - does not affect print speed

radek8, 2020-08-03:
look at
#680 
#757

radek8, 2020-08-03:
v Marlin turns off the automatic return to the home screen
#define LCD_TIMEOUT_TO_STATUS -1
Then switch to the menu in Marlin, then it'll be okay
Be careful not to click on the encoder, otherwise you will move in the menu on the Marlin screen

radek8, 2020-08-03:
Or insert the HW switch on the EXP2 cable :-)

Tygrys-1, 2020-08-03:
> 
> 
> Or insert the HW switch on the EXP2 cable :-)

Or EXP3 in this hardware case. :-)
But maybe we can modify Marlin to ignore the encoder under some condition? Something like a "console lock" to prevent children to play with the oven?

Tygrys-1, 2020-08-03:
> 
> Be careful not to click on the encoder, otherwise you will move in the menu on the Marlin screen

And start a printing from SD. Or heat the nozzle to 250C. Or - who knows what else? ;-)

oldman4U, 2020-09-03:
Hi.

Please close this ticket

closed 2020-09-04 by NicoletoAerials

radek8, 2020-12-09:
Guruathwal has created a fix to address this issue.
the condition is a new FW Marlin and connection via connectors EXP1 and EXP2
You have an EXP3 connection, in which case this adjustment does not work.

I have an idea how to apply Guruathwal repair also for SKR mini E3 board with EXP3 connector. It means a small modification of Marlin and the connection of one wire.
Are you interested in trying it out?