bigtreetech/BIGTREETECH-TouchScreenFirmware #890: [Q] POWER_LOSS_RECOVERY

DrumClock, 2020-07-30:
`POWER_LOSS_RECOVERY` does not work after replacing LCD 12864 with TFT 50 Touch screen .
SD card from which I am printing is in the touch screen.

**Where should I look for a bug?**

radek8, 2020-07-30:
Did you compile the FW with #876 and set the ini file?

DrumClock, 2020-07-30:
Hi, power failure information is not connected to the TFT 50 but to the RUMBA board.

Tygrys-1, 2020-07-31:
There might be some misconceptions when using features of different boards. When printing from TFT, the TFT is not aware of systems connected to the mainboard. When printing from the mainboard, the Marlin is not aware of sensors connected to the TFT. 
POWER_LOSS_RECOWERY works by writting the current layer information on SD, but as the SD card is not inserted at the MB's slot, it will do nothing in case of printing from SD card inserted in TFT. So the responsibility to handle the power loss is then on the TFT. If you run the print from the TFT, it uses it's own protection, which is not driven by POWER_LOSS_RECOVERY. Might be what @radek8 suggested.

radek8, 2020-07-31:
What's this?

[image]

radek8, 2020-07-31:
That's not it
[image]

bigtreetech, 2020-07-31:
Hello, are you using BTT Mini UPS module? If so, please set Marlin according to the manual.
and the touch screen only needs to be set according to the following figure in config.ini file
[image]

DrumClock, 2020-07-31:
Unfortunately, if the SD card is in the TFT 50 board,
 then `POWER_LOSS_RECOVERY` in Marlin does not work.

bigtreetech, 2020-07-31:
the Z-axis will not raised when power loss?

DrumClock, 2020-07-31:
NO, if the SD card is in the FTF board, then the function is not active in Marlin

bigtreetech, 2020-07-31:
Which motherboard are you using? Just enable `#define SDSUPPORT` in the firmware, and then set the parameters of PLR. Print TFT SD on the touch screen will work normally.

DrumClock, 2020-07-31:
I am using a RUMBA mega2560 board in FW Marlin, `#define SDSUPPORT `is enabled and `PLR parameters` are set.
If I use old LCD 12864 controller (EXP1 and EXP2) everything is fine.

When using TFT 50 V3.0 (RS232) it does not work.
**The Rumba and TFT board doesn't know what the other is doing and that will probably be the whole problem.**

bigtreetech, 2020-07-31:
As far as I know, the logic of Marlin for PLR is that whenever a power loss is detected, the z-axis will be raised, whether Marlin is currently in printing or not. 
So if you can enable #define SDSUPPORT and set  PLR parameters in the marlin firmware.  it should work.
in other word, the firmware of PLR worked in LCD12864 does not need to be changed. You just need to unplug LCD12864. and plug in TFT50, set TFT50 in config.ini for PLR, and print TFT SD.  it should work.

DrumClock, 2020-07-31:
> As far as I know, the logic of Marlin for PLR is that whenever a power loss is detected, the z-axis will be raised, whether Marlin is currently in printing or not.

Marlin was changed.
If it does not print, PLR will not occur in the event of a power failure.

bigtreetech, 2020-08-01:
This does cause problems! You can use the previous version of Marlin to use this function, or wait for us to submit PR to Marlin to add the gcode command that tells Marlin to start printing.

DrumClock, 2020-08-01:
> You can use the previous version of Marlin to use this function,

### This is the same as using the LCD 12864 and not using the TFT 50 - step back.
The problems must be solved on the side of the touch screen and communication with Marlin.

**I tried this:**
1) I connected the old LCD 12864 with SD card to the RUMBA board using EXP1 and EXP2
2) I connected TFT 50 V3.0 to the RUMBA board using RS232
3) from the touch screen I start printing from the SD card which is in the LCD
4) `POWER LOSS RECOVERY `occurred during a power failure
5) unfortunately, after resetting, the TFT display did not offer the **RESUME PRINT** or **STOP PRINT** option.

So it probably doesn't work properly either.

DrumClock, 2020-08-03:
I thought of using the **PS-ON connector** and allowing it to be used to detect `PLR `from the UPS because the FTF 50 does not have its own PLR connector.

**Isn't this the fundamental problem?**
```
# ------------------------------------------------- -------------------
# Power Loss Recovery & BTT UPS Settings (if connected to TFT Controller)
# ------------------------------------------------- -------------------
```

Tygrys-1, 2020-08-03:
IMHO PLR has nothing to do with the UPS hardware. It just saves file pointer position information on the SD card and in case of power failure the temporary file stays on the SD, So on next startup it will find the file, read the position information and resume printing from that point. If the print finishes ok, the file will be deleted. Having UPS might preserve more exact position information when the UPS signal the power failure and do a smooth shutdown, moving the head out of printing model and so. On restart the print should resume from the point it was stopped.

Tygrys-1, 2020-08-03:
> 
> 
> > You can use the previous version of Marlin to use this function,
> 
> ### This is the same as using the LCD 12864 and not using the TFT 50 - step back.
> 
> The problems must be solved on the side of the touch screen and communication with Marlin.
> 
> **I tried this:**
> 
>     1. I connected the old LCD 12864 with SD card to the RUMBA board using EXP1 and EXP2
> 
>     2. I connected TFT 50 V3.0 to the RUMBA board using RS232
> 
>     3. from the touch screen I start printing "Onboard SD" from the SD card which is in the LCD
> 
>     4. `POWER LOSS RECOVERY `occurred during a power failure
> 
>     5. unfortunately, after resetting, the TFT display did not offer the **RESUME PRINT** or **STOP PRINT** option.
> 
> 
> So it probably doesn't work properly either.

The PLR will work in these cases: 
- you start the print from the Marlin interface from the "Onboard SD". Then, the PLR will be handled by the Marlin.
- you start the print from the TFT interface, from the TFT SD or USB. Then, the PLR will be handled by the TFT.

If you start the print from TFT, but using the onboard SD - then only the Marlin can handle the PLR, TFT only sends the command to print the file, and is not aware on details. It is the same, as if you started the print from the remote terminal. I'm not sure, how Marlin handles such a case.

DrumClock, 2020-08-04:
Hi, @Tygrys-1 and @bigtreetech 

> * you start the print from the Marlin interface from the "Onboard SD". Then, the PLR will be handled by the Marlin.

- **YES**, the PLR will process Marlin based on the information from the UPS `#define POWER_LOSS_PIN ` 
      does not occur after restart to resume print, because the TFT display does not receive any information from Marlin.

> * you start the print from the TFT interface, from the TFT SD or USB. Then, the PLR will be handled by the TFT.

- **NO**, the PLR cannot be processed because Marlin does not know that it is being printed from the SD card
    and does not  respond to the `POWER_LOSS_PIN` status change. 
     Because no information from "POWER_LOSS_PIN" is fed to the TFT display, nothing happens.

### **Therefore, setting as described by @bigtreetech above will never work**. 
 The only option I see is that the **PS-ON connector** could also be used for `POWER_LOSS_PIN` information from the UPS.

As you know, the TFT 50 V3.0 board does not have a separate PLR-info connector.
I don't know how fast "Marlin" will respond to a request to establish communication
 between the TFT and the board based on a G-code.

bigtreetech, 2020-08-13:
> > As far as I know, the logic of Marlin for PLR is that whenever a power loss is detected, the z-axis will be raised, whether Marlin is currently in printing or not.
> 
> Marlin was changed.
> If it does not print, PLR will not occur in the event of a power failure.
> 
> [MarlinFirmware/Marlin@9b504d4](https://github.com/MarlinFirmware/Marlin/commit/9b504d49ec30d42fc8d86d4d337e750bc163667c)

Hello, There are similar Gcode in Marlin. You just need to add M75 to start gcode and M77 to end gocde in the config.ini file of the touch screen. Just like the picture below. It should work.
[image]

DrumClock, 2020-08-13:
Yes from M75 PLR already works when printing from "TFT SD"
After power recovery, it would be advisable to start on the TFT menu
as with the LCD 12864.

[image]

closed 2020-08-19 by bigtreetech

Jedrusia, 2020-12-03:
It's not working with me