bigtreetech/BIGTREETECH-TouchScreenFirmware #892: [FR] (Add menu to power off printer after print)

xhejtman, 2020-07-31:
Add menu option so you can select **during print** that printer should power off after print (M81 cmd).

radek8, 2020-07-31:
You can paste the M81 command into an ini file

# end_gcode:G90\nG1 E-4\nG92 E0\nM18\nM81\

xhejtman, 2020-07-31:
But can I turn on/off the end_gcode while printing?

radek8, 2020-07-31:
Why?
So set it to fill in after each print.
You turn the printer back on before printing again
If you do not want to turn it off before printing, you will cancel it

xhejtman, 2020-07-31:
Well, the FR is about to select during print, whether the printer should turn off or not.

Also, you should wait until hotend is cool, not turn off with hot hotend and I'm not sure, you can actually cancel print while marling is waiting for sth.

radek8, 2020-07-31:
You can set at what hotend temperature the hotend cooling fan should turn off. Default 50 ° C. I assume the printer could turn off along with the fan.

radek8, 2020-07-31:
#define AUTO_SHUT_DOWN_MAXTEMP 50

bigtreetech, 2020-08-01:
As mentioned above, this function is already available in firmware. you can click "More > Feature > Auto power" to ON/OFF power off function in Printing

bigtreetech, 2020-08-03:
Hello， I will close here because this feature already exists, please confirm. If you have any questions, please reopen here and contact me

closed 2020-08-03 by bigtreetech

xhejtman, 2020-08-03:
Yes, it is already present, thanks.