bigtreetech/BIGTREETECH-TouchScreenFirmware #893: [BUG] (MARLIN 2.5.4 - M997 freeze SKR1.4 Turbo with BTT TFT35-E3 V3.0 only in touch mode)

jjtronics, 2020-07-31:
### Description

Hello, i have a BTT SKR1.4 Turbo witth BTT TFT35-E3 V3.0.
The Marlin firmware is the last 2.0.5.4.
If i send GCODE M997(for reboot) via seriab USB with Octoprint and the touchscreen is on TouchUI mode, le printer freeze (Serial port down and impossible to reconnect the printer).
If i change TouchUI mode to Marlin mode, the screen show nothing. 

If i send GCODE M997 (for reboot) and the touchscreen is on Marlin mode, le printer reboot propely, all is good.

### Steps to reproduce

1. Switch the screen to TouchUI mode
2. Send gcode command : M997 on usb serial port
3. The board is frozen.

**Expected behavior**
The M997 command should behave identically in Marlin mode as in TouchUI mode

**Actual behavior**
M997 command freeze SKR board when the display is on TouchUI mode

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details
current TFT Firmware version number : BIGTREE_TFT35_V3.0_E3.26.x.bin & Marlin firmware 2.0.5.4

Thank you in advance for you help !!!

jjtronics, 2020-07-31:
I have the same problem with Marlin 2.0.6

radek8, 2020-07-31:
try setting a value in the ini file
serial_always_on: 1

jjtronics, 2020-07-31:
I set serial_always_on: 1 (I took the opportunity to activate the led of the potentiometer, that's cool!)
But nothing change, marlin crash and i need to power off/on the power supplie :(

radek8, 2020-07-31:
My configuration
SERIAL_PORT - TFT display
SERIAL_PORT2 - wifi
if I send command M997 to SERIAL_PORT - restart OK
if I send the M997 command to SERIAL_PORT2 - the Marlin board will freeze

The problem will be in Marlin
use SERIAL_PORT to send an M997 command

jjtronics, 2020-08-01:
Hum, i dont think it's Marlin, my configuration : 
SERIAL_PORT = USB Serial (Octoprint)
SERIAL_PORT_2 = TFT35
SERIAL_PORT_3 = ESP3D

il send my gcode command only with octoprint on SERIAL_PORT
The bug depends on the mode chosen on the screen (In Marlin mode all is good, but in TouchUI mode Marlin freeze ...)

radek8, 2020-08-01:
> SERIAL_PORT = USB Serial (Octoprint)
> SERIAL_PORT_2 = TFT35
> SERIAL_PORT_3 = ESP3D

Marlin má SERIAL_PORT_3 ??
Jste si jistý?

jjtronics, 2020-08-04:
Yes

radek8, 2020-08-04:
You will probably be the only one
I have downloaded a 6 hour old Marlin bugfix-2.0.x version, but I don't see this option there ...

testack18, 2020-08-05:
I have the same issue. Some additional notes to the above:
If I switch to the graphics UI on the TFT and enter to the Terminal and send gcode M997 it looks nothing happens. But immediately it drops the connection with Octoprint on USB. Meanwhile Octoprint sais:
SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)'

I just have downloaded and flushed the latest Marling 2.0 bugfix and also the lastest BTT TFT35 firmware.

jjtronics, 2020-08-06:
I'm not alone !

blueeagle69, 2020-08-16:
Same problem on my Biqu  B1 (SKR1.4 and TFT35 E3).

testack18, 2020-09-01:
Well, I have updated the TFT firmware and also I have installed the latest marlin bugfix. Since the latest TFT now my motherboard freezes also on the emulated screen too. On the graphics screen simple looks like does nothing but on the emulated screen M997 is able to reboot the board but it freez and I manually have to reset or power-cycle the printer.

Any update on this? Anyone else experiencing the same?

jjtronics, 2020-09-05:
We are several to have the same problem, I even think that all have this problem but that few do reset unfortunately ... So it does not interest anyone

DodgeDeBoulet, 2020-11-04:
Thank you for linking to my issue, @oldman4U. Doesn't seem as though anyone's gotten to the root cause yet :(

oldman4U, 2020-11-04:
Seems so;-(

But I added this issue to the centralized bug list, so hopefully a
developer will take care one day.

You can close the ticket if you want.

Happy printing

DodgeDeBoulet, 2020-11-04:
My only reluctance in closing the ticket is that it contains additional forensic information; for example, that the board does actually reset and flash firmware if present on SD card, but no longer responds to USB after the firmware is flashed.

oldman4U, 2020-11-04:
Paste all important information and a link to your ticket here. All information is important.

DodgeDeBoulet, 2020-11-04:
[Link to identical issue I posted on the SKR 1.3/1.4 issues list](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/447).

I'm having an issue with a BTT SKR 1.4 Turbo and TFT35 v3.0. Issuing an M997 "reset" command when the display is in Graphic/TFT mode will reset the board, but it does not re-establish USB connectivity to the host; /dev/ttyACM0 does not become available again. In "Marlin" mode the reset works normally.

I've running Marlin 2.0.7.2, and have tried 2.0.7.1 and 2.0.6.1 in an attempt to determine if Marlin is the culprit, but the issue happens on all 3 versions.

Note that if there is a firmware.bin file on the SD card, the M997 does result in it being flashed. But after the flash completes, USB connectivity to the host (in my case a Raspberry Pi 4) is not re-established.

A power cycle of the board returns operation to normal and /dev/ttyACM0 becomes available again.

oldman4U, 2020-11-05:
Thank you DodgeDeBoulet.

Happy printing

closed 2021-01-11 by stale[bot]

oldman4U, 2021-01-14:
Hi jjtronics.

Could you please be so kind and check, if the current master firmware solves the reported issue.

According to the responsible developer, #1468 should solve it.

Thank you

oldman4U, 2021-01-15:
Hi again.

I just verified that M997 does not cause any issues any more when in touch mode, so I remove this issue from the Bug list.

Happy printing