bigtreetech/BIGTREETECH-TouchScreenFirmware #895: [BUG] Custom  G-Code Commands

DrumClock, 2020-08-01:
### not working

traffic-light, 2020-08-02:
I've had the same issue.
See #898 for the solution

DrumClock, 2020-08-02:
But this menu does not work for me.
Every time I select something, the TFT freezes and I have to restart the printer

```

#--------------------------------------------------------------------
# Custom G-Code Commands (up to 15 custom G-code)
#--------------------------------------------------------------------

#### To enable a custom Command:
# Remove '#' at the begining custom commands label & gcode.

#### To disable a custom Command:
# Add '#' at the begining custom commands label & gcode.

# if the values are left blank then default name and G-code will be used

# Options: [Label maximum length: 24 characters, G-code Maximum length 70 characters]

custom_label_1:Servo central position
custom_gcode_1:M280 P0 S90\n
custom_label_2:Servo E1 position
custom_gcode_2:T0\n
custom_label_3:Servo E2 position
custom_gcode_3:T1\n
custom_label_4:End after G28 !
custom_gcode_4:G1 X300 F3000\nG1 Y300\nM280 P0 S90\nM84\nG1 Z10 F3000\n
custom_label_5:Save to EEPROM
custom_gcode_5:M500\n
custom_label_6:Restore from EEPROM
custom_gcode_6:M501\n
custom_label_7:EEPROM defaults
custom_gcode_7:M502\n
custom_label_8:Init sd card
custom_gcode_8:M21\n
custom_label_9:Release sd card
custom_gcode_9:M22\n
custom_label_10:restore leveling
custom_gcode_10:M420 S1\n
custom_label_11:Disable steppers
custom_gcode_11:M84\n
#custom_label_12:custom12
#custom_gcode_12:M105\n
#custom_label_13:custom13
#custom_gcode_13:M105\n
#custom_label_14:custom14
#custom_gcode_14:M105\n
#custom_label_15:custom15
#custom_gcode_15:M105\n
```

radek8, 2020-08-02:
Undo changes in #891?
MachineSetings.c file

[image]

bigtreetech, 2020-08-03:
Thanks for your feedback,  and this problem should be fixed in #904, please confirm.

DrumClock, 2020-08-03:
Everything is okay now

closed 2020-08-03 by DrumClock