bigtreetech/BIGTREETECH-TouchScreenFirmware #896: Folder upload

AntoszHUN, 2020-08-01:
How do I upload a folder?

oldman4U, 2020-10-11:
Hi.

I just make a new ticket with all the feature requests, so it is easier to search and also for the developers to see what has been requested. Unfortunately I do not understand your request, so could you please explain.

Thank you

AntoszHUN, 2020-10-11:
I wanted to create a folder for new menu icons. Only ZIP folder was created / uploaded. You can see it in the main directory.

Hello, AntoszHUN

oldman4U, 2020-10-13:
You mean while updating the firmware for example?

This is a folder and not a zip.

AntoszHUN, 2020-10-13:
[image]

I'm talking about this. I wanted to upload it as a folder. Only ZiP allowed.

oldman4U, 2020-10-16:
So you uploaded this theme, but it did not accept a folder, so you uploaded a zip, right?

I guess you did this using a PR?

AntoszHUN, 2020-10-16:
Yes. Exactly. Shouldn't there have been an enhancement then?

oldman4U, 2020-10-19:
Hi. Could you please check if this is what you want? https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1187

We should also update the ReadMe to cover this theme. Have you made it?

oldman4U, 2020-10-20:
I do not know which tools you are using, but  I use GitDesktop on Mac and there I had to add a comment before I could add the folder to my branch. Maybe this helps.

Could you please be so kind and close this ticket.

Jonapot

PR #1251 (2020-11-27): Changed the ReadMe to cover the new language.ini, added "The Round Miracle" and the "Hybrid Red" themes and removed the "Classic" theme
The new language.ini works different than it is described in the ReadMe today, which could confuse the user.
Added information about the cabling, because there are some users which do not understand how many cables to connect.

I removed the Classic menu, because it is not supported by the last firmware and added two themes which are supporting all TFTs from 24 to 70.

resolves #896 
resolves #1175
closed 2020-10-28 by bigtreetech