bigtreetech/BIGTREETECH-TouchScreenFirmware #897: USB printing has pauses

FacusFPV, 2020-08-01:
Good morning, upgrade my BIGTREE_TFT35 screen from version Vx.x.25 to Vx.x.26. Printing has pauses all the time, the only change was the fw. With the Vx.x.25 it works without problems, this only goes through usb from the sd reader of the screen does not. Try SERIAL_PORT-1 and 1, SERIAL_PORT_2 -1 and 1 and BAUDRATE 115200 and 250000

traffic-light, 2020-08-01:
I've got the same problem.
It always appears at a layer change, and sometimes randomly in the same layer, when printing from USB.

Printing from sd-card has no problems.

It would be great if it is possible to print normal from USB again.

I did a build with the latest version (with a lot of great add-ons) but this problem is still there
(I use 115200 baud, serial alway's on)

bigtreetech, 2020-08-04:
Hello, Can you compile and update the latest version of firmware? Then turn off the PLR feature in "Settings->feature->power loss recovery->OFF". and printing again?
In my test,  Only when the z-axis coordinates are changing very frequently in printing, which causes the file to be written to the U disk very frequently, and then there will be a delay. If the PLR feature is turned off, the write function will be disabled, and then the printing will become OK.

traffic-light, 2020-08-04:
Thanks for the reply. I will try this.

Another thing I've noticed is that the quality is somewhat better.
First I'll take a look with you're suggestions.

If quality isn't improved i'll reply

bigtreetech, 2020-08-04:
Thank you!
And can you provide the gcode file for your test so that I can test with the same environment

traffic-light, 2020-08-04:
@bigtreetech I didn't see any breaks/ pauses anymore. The print is also looking better now.
Thanks.

bigtreetech, 2020-08-05:
@traffic-light Hello, Is your model printed in "vase mode" ？

traffic-light, 2020-08-05:
No, I did a stringing test with 2 pillars and a 3dbenchy. Both without pauses

traffic-light, 2020-08-13:
In my opinion this one can be closed @FacusFPV @bigtreetech

closed 2020-08-13 by bigtreetech

Sebazzz, 2020-09-30:
> The print is also looking better now.

@traffic-light Which suggestion helped?

traffic-light, 2020-09-30:
> > The print is also looking better now.
> 
> @traffic-light Which suggestion helped?

"Settings->feature->power loss recovery->OFF"

piotrkochan, 2020-12-16:
In my case power loss recovery is not enabled in the Marlin, no option on the screen to toggle but still blobs are present while printing from the pendrive.