bigtreetech/BIGTREETECH-TouchScreenFirmware #898: [BUG] cancel gcode

traffic-light, 2020-08-01:
### Description

Cancel  gcode setting doesn't work. There is also no conection with the mainboard anymore.
The losing connection part is for so far only noticed by no axis movement possible

### Steps to reproduce

1. Start a print
2. Cancel the print
3. Print stops, temperature stay's the same and no connection with the mainboard

**Expected behavior**
Custom end gcode working

**Actual behavior**
End gcode doesn't do a thing and it looks like mainboard crashes

### Hardware Variant
BIGTREE_TFT35_E3_V3_0
### TFT Firmware Version & Main Board Firmware details
TFT: Latest version at the moment (.26 1 aug version)
Marlin: 2.0.5.3

### Additional Information
With the original and modified cancel end gcode it's not working

**Original**
`#### Cancel G-code - run this G-code after canceling print
# maximum length 50 characters
cancel_gcode:M104 S0\nM140 S0\nG28 X Y R10\n`

The end gcode is also activated:
'#### Default Cancel Gcode Status
# Options: [enable: 1, disable: 0]
cancel_gcode_enabled:1'

bryan065, 2020-08-02:
I'm having the same issues now as of Aug 1st

traffic-light, 2020-08-02:
Thank you. I will give it a try

traffic-light, 2020-08-02:
@bryan065 It works. Thank you.

I modified line 116 - 136 with your suggestion:
````
//send print codes [0: start gcode, 1: end gcode 2: cancel gcode]
void sendPrintCodes(uint8_t index)
{
  PRINT_GCODES printcodes;
  W25Qxx_ReadBuffer((uint8_t *)&printcodes, PRINT_GCODES_ADDR, sizeof(PRINT_GCODES));
  switch (index)
  {
  case 0:
    mustStoreCmd(printcodes.start_gcode);
    break;
  case 1:
    mustStoreCmd(printcodes.end_gcode);
    break;
  case 2:
    mustStoreCmd(printcodes.cancel_gcode);
    break;

  default:
    break;
  }
}
````

bigtreetech, 2020-08-03:
Thanks for your feedback,  and this problem should be fixed in #904, please confirm.

traffic-light, 2020-08-03:
Thank you for this quick bug fix. It works like it should be working :smile:

closed 2020-08-03 by traffic-light