bigtreetech/BIGTREETECH-TouchScreenFirmware #906: [BUG] (tft70 and skr mini e3 v2.0 no marlin)

Robenc, 2020-08-03:
LCD 12864 Simulator but no marlin

TFT 70 v3.0 and the skr mini e3 v2.0

BIGTREE_TFT70_V3.0.26.x.bin and skr mini is the firmware.bin listed on here no extras added to the board at this time

need help getting this fixed

oldman4U, 2020-08-03:
Hi.

LCD and Marlin mode is the same, so I guess you mean you have no touch screen mode?

How many and which cables have you connected between the display and the mainboard and which baud rate have you defined? How many and which serial ports do you have defined in the firmware of the mainboard? What exactly do you mean with no Marlin - maybe a picture can help?

Tygrys-1, 2020-08-03:
Just press the knob. The Marlin interface should show up.

Robenc, 2020-08-03:
it does not work

oldman4U, 2020-08-03:
Could you make a photo which shows the problem

oldman4U, 2020-08-03:
And maybe you can answer the question above. Thank you

Robenc, 2020-08-04:
found the problem hooked the data cable from exp1 to exp3 and now it is working

closed 2020-08-04 by Robenc

oldman4U, 2020-08-04:
Cables - an evergreen. 

Glad we could help