bigtreetech/BIGTREETECH-TouchScreenFirmware #907: Smart FIlament Sensor problem

AntoszHUN, 2020-08-04:
The Smart Filament sensor does not work when connected to the appropriate port on the display. It has been set in firmware. But there is no reaction. The thread jams. Printing goes on. There would always be no sensor. Does anyone know a solution? Maybe I didn't make any adjustments? I also set everything up in principle in Marlin.

Tygrys-1, 2020-08-05:
If you connect the sensor to the TFT and configure the right type in config.ini, then it wol work only when printing from the TFT SD card or USB.

lumenmesh, 2020-08-05:
I have my BTT smart sensor plugged into the back of the TFT35 v3.   New firmware upload, config.ini changed to 2 "smart".      Settings on the touchscreen set filament sensor to "smart".  Printing from SD card.    I hold the extruder open so no filament is being pushed through and nothing happens, so the sensor isn't working.    Any suggestions?     Do i need a new sensor?    Is there more in the conifg file i need adjust?    Seems like the compatibility between a BTT touchscreen and a BTT smart sensor is not as seamless as it could be....

AntoszHUN, 2020-08-05:
That is my remark for me too. The FIL-DET connector on the display is completely unnecessary. It only works when connected to the motherboard. But it's also only in Marlin mode. That’s exactly the point of forgetting marlin mode, because there’s so much going on from the touch screen.

AntoszHUN, 2020-10-27:
Why close? It has not been solved. Or yes? I do not know about it.

abecaw, 2020-12-17:
I have it working in touch screen mode as well as simulation mode and optioning. I have a Ender 3 pro with the BTT SKE mini E-3 V-1.2 mainboard. I have the TFT35 E3 V3.0. I have a Micro Swiss Direct Drive all metal hot end. Oh and a BlTouch. I'm using the latest version of Kay3D Cheetah and Marlin 2.0.7.2. I'm using PT-DET connector.

AntoszHUN, 2020-12-17:
> I have it working in touch screen mode as well as simulation mode and optioning. I have a Ender 3 pro with the BTT SKE mini E-3 V-1.2 mainboard. I have the TFT35 E3 V3.0. I have a Micro Swiss Direct [...]

I in the BTT smart filament sensor thought something was wrong. Does not work.

killrob, 2021-01-24:
some news? i have the same problem, connected the smart filament sensor on fil-det in the back of TFT35 E3 V3 and nothing happens when the filament runs out. I made a try opening the spring on the extruder so filament cant be pushed and nothing happens

oldman4U, 2021-03-21:
HI.

FYI
I am going to start a test very soon including the BTT Smart Filament Sensor and a TFT35 E3 v3 running the current Marlin and TFT firmware. Is there anything what you could let me know in addition to what is written here in this ticket?

Thank you

oldman

killrob, 2021-03-22:
@oldman4U at the end my smart filament sensor works in this way:
1. connected to the back FIL-DET port of the TFT35
2. using the last TFT35 E3V3.0 firmware
3. set in the right way in the config.ini (fil_runout:2 - fil_runout_inverting:1 - fil_noise_threshold:100 - fil_runout_distance:7)
4. disabled LIN_ADVANCE in Configuration_adv.h
5. de-commented M114_DETAIL always in Configuration_adv.h 
Using Marlin 2.0.7.2
i'm using TFT35 only in touch mode, with this configuration in Marlin emulation mode doesn't works

AntoszHUN, 2021-03-22:
> @oldman4U at the end my smart filament sensor works in this way:
>
>     1. connected to the back FIL-DET port of the TFT35
>
>     2. using the last TFT35 E3V3.0 firmware
>
>     3. set in the right way in [...]

I'll try.

oldman4U, 2021-03-22:
Hi.

I have been testing standard filament sensor with printing from TFT storage and it worked perfectly. Same configuration but printing from on board SD worked also fine, but resume sometimes took 2-3 tries until it really resumed. Higher baud rates worked better, lower baud rates "repeated" more likely and more often.

Linear Advance caused so many problems on my printer that I turned it off finally.

So I will verify your result and hope AntoszHUN can report the same.

Marlin Emulation I did not try but I expect Marlin will work or not. There is not a lot this FW can do, like in situations where the sensor is connected to the mainboard.

Happy testing

killrob, 2021-03-22:
rigth, at this moment i print only from TFT SD or USB, still not tried from SD onboard, i will try also this way.
hope that you and @AntoszHUN will meke it work as i've done.
If not tell me i will try to help ;)

oldman4U, 2021-03-23:
Hi killtob

Thank you for the update. I have been testing the smart filament sensor connected to the TFT using the Touchscreen mode printing from TFT SD CARD and USB as well as on board SD and it works very well. 100% when printing from TFT, 90% from on board perfectly and sometimes resume happens 2-3 times and not at the first time.

oldman4U, 2021-03-23:
I will remove the filament related tickets from the verified bugs list one digant73 PR is released.

killrob, 2021-03-23:
> Hi killtob
>
> Thank you for the update. I have been testing the smart filament sensor connected to the TFT using the Touchscreen mode printing from TFT SD CARD and USB as well as on board SD and it [...]

good to know, hope that also @AntoszHUN will reply like you :) and solved his issue

oldman4U, 2021-03-23:
Yes. This would be great.

👍🏻

Thank you again for your help

killrob, 2021-03-23:
we were here to help if possible ;)

killrob, 2021-03-23:
well i modified my Ender 3 with:
SKR mini E3 V2.0
TFT35 E3 V3.0
BTT Smart filament sensor
Marlin 2.0.7.2

oldman4U, 2021-03-23:
Ender 3 Pro

SKR E3 DIP 2208
Same TFT
No filament sensor installed

killrob, 2021-03-23:
is with this config and hardware that i made smart filament work ;)

AntoszHUN, 2021-03-23:
Unfortunately, I haven't had time to test yet. I hope I can test it tomorrow night. I would even ask if it matters tft or is there a smart filament sensor in the motherboard connector?

killrob, 2021-03-23:
not in the SKR Mini E3 V2.0

oldman4U, 2021-03-23:
You should do less Marlin translations😁

This ticket is about connected to TFT and that it does not work at all. You know that connecting to Mainboard Marlin takes over control and that Marlin does not care about others which are connected.

digant73, 2021-03-23:
> @oldman4U at the end my smart filament sensor works in this way:
>
> 1. connected to the back FIL-DET port of the TFT35
> 2. using the last TFT35 E3V3.0 firmware
> 3. set in the right way in the config.ini [...]

Can you test my PR #1752 and provide me the result if it woks also with linear advance enabled in marlin? You must enable M114_REALTIME in marlin. Thanks

killrob, 2021-03-23:
> Can you test my PR #1752 and provide me the result if it woks also with linear advance enabled in marlin? You must enable M114_REALTIME in marlin. Thanks

hi @digant73 
i've already done this test before posting here my working configuration, and i can tell you that with Linear Advance enabled and M114_REALTIME enabled in Marlin the SFS don't work at all. It works with Linear advance disabled and M114 enabled.

AntoszHUN, 2021-03-24:
Yes Yes. Less Marlin translation. But I don't really use marlin either. My next question is also here. What if I want to use 2 smart filament sensors? I have a 2in1 hotend with 2 extruders.

oldman4U, 2021-03-24:
So the first does not work but you want to use two of them!?

AntoszHUN, 2021-03-24:
> So the first does not work but you want to use two of them!?

I just asked what if I wanted to use two. Where do I connect the second one?

oldman4U, 2021-03-24:
So your problem with the filament sensor is not that it does not work but that you can not connect two of them. For me the ticket was like the first one does not work.

digant73, 2021-03-24:
> hi @digant73
> i've already done this test before posting here my working configuration, and i can tell you that with Linear Advance enabled and M114_REALTIME enabled in Marlin the SFS don't work at [...]

thanks for the test. So basically you got the same behavior as with the current BTT fw, right? or did you get different behavior with LA enable and disabled compared to BTT fw behavior? I was expecting my PR solved the problem when LA is enabled in Marlin and it continue to work also when LA is disabled

AntoszHUN, 2021-03-24:
The problem is that one doesn't work. In the meantime, I dropped the topic. Since then, I've rebuilt the machine. I didn't play smart filament sensor. There have been two extruders since then. I haven't tried the fix yet. But if it works, I would like another filament sensor for the other extruder. Incidentally, I raised the issue of the dual extruder sensor.

oldman4U, 2021-03-24:
But this ticket is about one which does not work. So do you want to test it
if it works or not?

AntoszHUN, 2021-03-24:
> […](#)

If I know, I'll test it today.

killrob, 2021-03-24:
> thanks for the test. So basically you got the same behavior as with the current BTT fw, right? or did you get different behavior with LA enable and disabled compared to BTT fw behavior? I was expecting my PR solved the problem when LA is enabled in Marlin and it continued to work also when LA is disabled

With LA enabled the SFS don't work as expected, the printer stops always telling filament was ran out

oldman4U, 2021-03-24:
I get a printer halted kill all with LA enabled independent of the available of a filament sensor SFS or not. But this ticket is ready about the SFS and if it works at all.

digant73, 2021-03-24:
> With LA enabled the SFS don't work as expected, the printer stops always telling filament was ran out

did you get this behavior with both my PR and BTT fw or only with my PR? Eventually which is the behaviour you get with BTT fw?

killrob, 2021-03-24:
> did you get this behavior with both my PR and BTT fw or only with my PR? Eventually which is the behaviour you get with BTT fw?

Sorry.... This behaviour is obtained with both, also with your PR.
If i enable LA i got SFS not working, with both.

AntoszHUN, 2021-03-24:
> @oldman4U at the end my smart filament sensor works in this way:
>
>     1. connected to the back FIL-DET port of the TFT35
>
>     2. using the last TFT35 E3V3.0 firmware
>
>     3. set in the right way in [...]

I tried it quickly. Parameters. BTT SKR V1.4 Turbo motherboard Marlin Bugfix-2.0.x and TFT70 display. Sensor connected to TFT70. It works based on the above settings! Thanks for the help. But what if you want to use Linear Advance?

killrob, 2021-03-24:
> I tried it quickly. Parameters. BTT SKR V1.4 Turbo motherboard Marlin Bugfix-2.0.x and TFT70 display. Sensor connected to TFT70. It works based on the above settings! Thanks for the help. But what [...]

i tried many times enabling LA, but in all cases the sensor stops working. And sincerely i did not investigate further, cause i'm satisfied with this goal achieved. I have this configuration working on all my 4 printers, two of them equipped with SKR mini E3 v2.0 and tft35 one equipped witk SKR V1.4 and tft35 and the last with SKR V1.4 and tft70.

AntoszHUN, 2021-03-24:
So where do I plug the second filament sensor?

oldman4U, 2021-03-24:
So you use LA on your printer?

I believe the TFT supports a single sensor. At least I never say anything about support for more. You just want to know this or are you using dual filament and LA.

Finally it is great to hear that the reason for this ticket does not exist anymore and that you have s setup now which you could use to help to make it work with LA and dual+ extruder!

Maybe you want to close this ticket and start two new tickets which cover the new topics.

AntoszHUN, 2021-03-24:
Yes, I will. I'm closing too. The error is resolved. Many thanks!

closed 2021-03-24 by AntoszHUN

oldman4U, 2021-03-25:
Köszönöm szepen

stefi01, 2021-09-09:
i pluged the sensor into fil runout pin on the btt mini e3 v2.0, configured how it should be but it dont work, keeps saying filament change and removes filament from the nozzle and then turns the sensor off in settings.

oldman4U, 2021-09-09:
How is this related to the firmware of the TFT?

AntoszHUN, 2021-09-09:
The SFS must be connected to the TFT FIL_DET connector.

stefi01, 2021-09-09:
> The SFS must be connected to the TFT FIL_DET connector.

thanks, i found the pin for it on the tft, changed the tft firmware config file and it finally worked :)

MartinM8266, 2023-01-26:
Hello, is there a way for Klipper with BKS Mini E3 V3.0 and TFT35 V3.0 ?
Klipper console isnt´t recognizing filament motion sensor when connected to PC15 (E0 endstop connecttor). thanks!

AntoszHUN, 2023-01-28:
> ID: <bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/907/1405822999@

Hello! I think the needle needs to be reversed or the code is missing a
notation.

Martin ***@***.***> ezt írta (időpont: 2023. jan. 27., P
0:48):