bigtreetech/BIGTREETECH-TouchScreenFirmware #915: Using an encoder in touch screen mode?

AntoszHUN, 2020-08-05:
A question arose as to whether this encoder could be used in touch screen mode? Can it be solved by programming? For example, when entering values or just navigating the menu. Can anyone be interested in this thing?

Hello, AntoszHUN

traffic-light, 2020-08-05:
It's already mentioned a lot. But when using the encoder in touch mode, it will also do things in marlin. 
It's hardwired through the flat cables. So, when you want to disable the encoder in marlin there should change something in marlin or in the hardware for the screen.

I'm thinking to make sort of plug witch you can break the connections to the band cable. Some sort of switch that can be controlled by the tft screen.

AntoszHUN, 2020-08-05:
Do not quite understand. But so far it has come down that it cannot be solved.

traffic-light, 2020-08-06:
I'm sorry. Maybe this will help:

look at
#680
#757

AnHardt, 2020-08-06:
Could sending the Marlin-Menu into a lock-screen, just preventing from leaving the lock-screen by the encoder, lockable and unlockable by g-codes when entering/leaving Marlin-Mode, be a solution?
https://github.com/MarlinFirmware/Marlin/issues/18661

traffic-light, 2020-08-06:
That could be helpful yes

radek8, 2020-08-06:
This is a beautiful solution, but modified by Marlin

radek8, 2020-08-07:
###  I have a temporary solution to this problem.
I set up in Marlin
#define LCD_TIMEOUT_TO_STATUS -1
This deactivates the return to the status screen in Marlin.
After starting the printer, press the encoder knob  once.
Marlin switches to the menu in the background. Returning to the status screen in Marlin is deactivated, so the menu remains open.
I can rotate the touch screen with the encoder and the print speed does not change. Wow

closed 2020-08-08 by AntoszHUN