bigtreetech/BIGTREETECH-TouchScreenFirmware #920: [BUG] ? Stuck on Black screen when switched to Marlin mode

charakaSamawry, 2020-08-06:
Description
Latest firmware from last night (master) and when switching from touchscreen mode to marlin mode it as usual goes black screen but the encoder whee doesnt 'bring it to life' now and restarting the printer, it stays stuck on the black screen, though i thought it should reboot into touchscreen mode as default.. (touchscreen mode works fine, but no mesh edit like on marlin mode) 

Steps to reproduce
1. Turn on touchscreen with SD card update
2. Change to marlin mode
3. Move encoder wheel

**Expected behavior**
Switches to marlin mode

**Actual behavior**
stuck on Black screen.

Hardware Variant
TFT35 V3 E3 with SKR mini v2 E3

TFT Firmware Version & Main Board Firmware details
Both flashed last night, master and bugfix.

Additional Information
https://github.com/charakaSamawry/BIGTREETECH-TouchScreenFirmware/tree/sam1

-https://github.com/charakaSamawry/BIGTREETECH-TouchScreenFirmware/blob/sam1/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme/config.ini

https://github.com/charakaSamawry/Marlin/tree/sam1

-https://github.com/charakaSamawry/Marlin/blob/sam1/Marlin/Configuration.h
-https://github.com/charakaSamawry/Marlin/blob/sam1/Marlin/Configuration_adv.h

traffic-light, 2020-08-06:
It could be a problem with my code in that latest commit.
I'm searching for a solution.

traffic-light, 2020-08-06:
That's why I din't noticed it.. 
It doesn't work in combination with #910 OR #912

traffic-light, 2020-08-06:
Could it be that you have activated the wrong version? instead of 12864 mode the 2004 mode?

charakaSamawry, 2020-08-07:
Sorry dont think you had commited your change when i found the bug, but have tried your fork anyway to no avail though sadly. Tried with default config as well incase id changed something weird... (and without custom theme just in case)
Problem still happening, switch to marlin mode black screen doesnt go away with move of the encoder wheel and stays black but lit even when restarted. Only fixed by a reflash. Reset button seems to make the back light flicker but nothing else..

I wasnt sure where you changed the settings you mentioned regarding 12864 and 2004 mode..?

traffic-light, 2020-08-07:
I think you've got an older config.ini.
But you did try it with the default config...

It's a strange error.. All cables are correct?

traffic-light, 2020-08-07:
Did you uncommanted this lin:  #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER ?
See https://github.com/bigtreetech/BIGTREETECH-TFT24-V1.1/issues/6

charakaSamawry, 2020-08-07:
Previously reprap_discount_full_graphic_smart_controller failed to compile for me, the error saying only 'these' are compatible with the skr mini board,and bad to put it back to cr10_stock_display to get it to compile and had worked before a more recent update.

digant73, 2020-08-07:
#933

charakaSamawry, 2020-08-07:
Preliminary testing, issues fixed. Will update when ive reinstated my personal configurations

charakaSamawry, 2020-08-07:
> You only have on cable and it worked before.. So this is not needed I think...
> 
> _Did you uncommented this lin: #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER ?
> It's not uncommented in one of your urls.
> See [bigtreetech/BIGTREETECH-TFT24-V1.1#6](https://github.com/bigtreetech/BIGTREETECH-TFT24-V1.1/issues/6)
> 
> This was, for me, the solution in marlin_

Just tried using reprap_discount_fullgraphic to no avail. Have always had to use cr10_stockdisplay to work with skr mini v2. 
This is the error message:
In file included from Marlin\src\HAL\STM32F1\../../inc/../pins/stm32f1/pins_BTT_SKR_MINI_E3_V2_0.h:33:0,
                 from Marlin\src\HAL\STM32F1\../../inc/../pins/pins.h:522,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/../pins/stm32f1/pins_BTT_SKR_MINI_E3_common.h:173:6: error: #error "Only CR10_STOCKDISPLAY, ZONESTAR_LCD, ENDER2_STOCKDISPLAY, and MKS_MINI_12864 are currently supported on the BIGTREE_SKR_MINI_E3."
     #error "Only CR10_STOCKDISPLAY, ZONESTAR_LCD, ENDER2_STOCKDISPLAY, and MKS_MINI_12864 are currently supported on the BIGTREE_SKR_MINI_E3."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/../pins/stm32f1/pins_BTT_SKR_MINI_E3_V2_0.h:33:0,
                 from Marlin\src\HAL\STM32F1\../../inc/../pins/pins.h:522,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:35:
Marlin\src\HAL\STM32F1\../../inc/../pins/stm32f1/pins_BTT_SKR_MINI_E3_common.h:173:6: error: #error "Only CR10_STOCKDISPLAY, ZONESTAR_LCD, ENDER2_STOCKDISPLAY, and MKS_MINI_12864 are currently supported on the BIGTREE_SKR_MINI_E3."
     #error "Only CR10_STOCKDISPLAY, ZONESTAR_LCD, ENDER2_STOCKDISPLAY, and MKS_MINI_12864 are currently supported on the BIGTREE_SKR_MINI_E3."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/../pins/stm32f1/pins_BTT_SKR_MINI_E3_V2_0.h:33:0,
                 from Marlin\src\HAL\STM32F1\../../inc/../pins/pins.h:522,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/../pins/stm32f1/pins_BTT_SKR_MINI_E3_common.h:173:6: error: #error "Only CR10_STOCKDISPLAY, ZONESTAR_LCD, ENDER2_STOCKDISPLAY, and MKS_MINI_12864 are currently supported on the BIGTREE_SKR_MINI_E3."
     #error "Only CR10_STOCKDISPLAY, ZONESTAR_LCD, ENDER2_STOCKDISPLAY, and MKS_MINI_12864 are currently supported on the BIGTREE_SKR_MINI_E3."
      ^~~~~
Compiling .pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\SoftwareSerial.cpp.o
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\HAL.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\HAL_SPI.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\Servo.cpp.o] Error 1

charakaSamawry, 2020-08-07:
> You only have on cable and it worked before.. So this is not needed I think...
>
> _Did you uncommented this lin: #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER ?
> It's not uncommented in one of [...]

Confused, as ive always got this error message when trying to use reprap as i see ther people configure for the BTT TFT35 V3
'
In file included from Marlin\src\HAL\STM32F1\../../inc/../pins/stm32f1/pins_BTT_SKR_MINI_E3_V2_0.h:33:0,
                 from Marlin\src\HAL\STM32F1\../../inc/../pins/pins.h:522,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/../pins/stm32f1/pins_BTT_SKR_MINI_E3_common.h:173:6: error: #error "Only CR10_STOCKDISPLAY, ZONESTAR_LCD, ENDER2_STOCKDISPLAY, and MKS_MINI_12864 are currently supported on the BIGTREE_SKR_MINI_E3."
     #error "Only CR10_STOCKDISPLAY, ZONESTAR_LCD, ENDER2_STOCKDISPLAY, and MKS_MINI_12864 are currently supported on the BIGTREE_SKR_MINI_E3."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/../pins/stm32f1/pins_BTT_SKR_MINI_E3_V2_0.h:33:0,
                 from Marlin\src\HAL\STM32F1\../../inc/../pins/pins.h:522,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:32,
Compiling .pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\SoftwareSerial.cpp.o
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:35:
Marlin\src\HAL\STM32F1\../../inc/../pins/stm32f1/pins_BTT_SKR_MINI_E3_common.h:173:6: error: #error "Only CR10_STOCKDISPLAY, ZONESTAR_LCD, ENDER2_STOCKDISPLAY, and MKS_MINI_12864 are currently supported on the BIGTREE_SKR_MINI_E3."
     #error "Only CR10_STOCKDISPLAY, ZONESTAR_LCD, ENDER2_STOCKDISPLAY, and MKS_MINI_12864 are currently supported on the BIGTREE_SKR_MINI_E3."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/../pins/stm32f1/pins_BTT_SKR_MINI_E3_V2_0.h:33:0,
                 from Marlin\src\HAL\STM32F1\../../inc/../pins/pins.h:522,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/../pins/stm32f1/pins_BTT_SKR_MINI_E3_common.h:173:6: error: #error "Only CR10_STOCKDISPLAY, ZONESTAR_LCD, ENDER2_STOCKDISPLAY, and MKS_MINI_12864 are currently supported on the BIGTREE_SKR_MINI_E3."
     #error "Only CR10_STOCKDISPLAY, ZONESTAR_LCD, ENDER2_STOCKDISPLAY, and MKS_MINI_12864 are currently supported on the BIGTREE_SKR_MINI_E3."
      ^~~~~
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\HAL.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\Servo.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\HAL_SPI.cpp.o] Error 1
`

digant73, 2020-08-08:
@charakaSamawry if I understood,Marlin mode was working on previous TFT fw while is not working with the last TFT fw. If so, try the solution on #933. That should fix your problem

digant73, 2020-08-08:
@charakaSamawry if I understood, Marlin mode was working with previous TFT fw and it is not working with the last TFT fw. If so try the solution on #933

closed 2020-08-12 by charakaSamawry