bigtreetech/BIGTREETECH-TouchScreenFirmware #921: G29 and other Gcodes in top of Marlin mode

RicardoEmePe, 2020-08-06:
### Description

When I switch to Marlin mode, at the windows  top some Gcodes are displayed. I had before firm. TFT35_V3.0.25.1 and  it works well !!!
[image]

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**
No Gcodes in top. 

**Actual behavior**
Read description bug

### Hardware Variant
BIGTREETECH_TFT35_V3.0
### TFT Firmware Version & Main Board Firmware details
TFT35_V3.0.26.x jul 13 2020  // SKR V1.3

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

guruathwal, 2020-08-07:
you need to reflash the config.ini file

RicardoEmePe, 2020-08-07:
Ok, now it's Ok. Yesterday eveningh I did it, but no changes. Now, Ok and "Invalid Keyword" message while reading config.ini don't show " Lcd12864_title: LCD12864 simulator" , and "lcd2004_title: LCD2004 simulator".
Anyway, while config.ini reading, the message in "Invalid Keyword" shows: 
ack_popup_type:1
marlin_type:1
bed_en:1
chamber_en:0
pl_recovery_en:1
knob_led_idle:1
lcd_brightness:11

I don't know if this is important. 

Thanks, guruathwal, for your quick reply !!!

guruathwal, 2020-08-11:
that's because you are using a new config file with old firmware

traffic-light, 2020-08-13:
@RicardoEmePe you can close this one when it's working again

RicardoEmePe, 2020-08-13:
I come back home next Sunday, I’ll try and close if ok. Many thanks

RicardoEmePe, 2020-08-17:
New firmware and new config file. Same problem. Enclosed you have a film with the  issues

[https://drive.google.com/file/d/1ProUmeQCrPmC9IbuyJh4kZ1BtTsY6WxV/view?usp=sharing]

traffic-light, 2020-08-18:
We don't have permission to see the file

closed 2020-09-01 by bigtreetech