bigtreetech/BIGTREETECH-TouchScreenFirmware #927: TFT24 freezes randomly

matbhe, 2020-08-07:
Hello

I'm using the TFT24 with a SKR 1.4 Turbo (marlin 2.0.x-bugfix) for a CNC machine. Because this machine has 0 extruders it does not know the M221 gcode. To fix the continuous error messages on the TFT24 i commented out the storeCmd("M221") in StatusScreen.c. After the upgrade the display works but randomly freezes (touch does no longer react to input). As soon as i comment out the other storeCmd - calls (M114/M220) in statuspage.c it seems to work without crashing. I could not find the bug until now. Does anyone have the same problems?

Regards Matthias

bigtreetech, 2020-08-10:
Hello, Can you confirm if the problem still exists after #950
Thank you very much!

matbhe, 2020-08-10:
Hello

The bug seems to be fixed with #950 . 
Well done :-)

Regards Matthias

matbhe, 2020-08-10:
Fixed

closed 2020-08-10 by matbhe