bigtreetech/BIGTREETECH-TouchScreenFirmware #932: IDEA: Gauges and progress on print screen

Estivio, 2020-08-07:
The screen when printing is not really glancable. It would be awesome if there were gauges and progress bars to visualize the current status. Like for print progress, and temperatures.
Apart from the obvious progress bar, I am imagining something that tells me how far my temps are away from the set value by a color or something visually similar to a boxplot.
If I have some spare time I will make a mockup on what I mean.

oldman4U, 2020-08-07:
Hi. You need someone who tells you what is possible and which hard and software limitations exist.

Estivio, 2020-08-07:
All right, I assume this is a polite way to say: This isn't possible? Or is it? 
I honestly did not look into the code at all. In my mind it was like: Get the target value and the current value and calculate the percentage an then draw a sprite with a color that is dependent on this value. (For a basic temperature gauge for example) I'm sorry, if it is simply not possible I close this issue immediately.

oldman4U, 2020-08-07:
Hi Estivio. 

It was not my intention to say that this is not possible or even that you should close the ticket. Sorry if you had this feeling after my posting. The opposite is the truth. I totally agree with you that the UI could be more fancy - and it was my pleasure to work with one of the developers here on a new interface for a few weeks some time ago. Unfortunately we have not been able to finish it, but it included many things like:

*Improved navigation
*"Monochrome" Interface where color has a function
*Additional navigation bar
*Additional Main Printscreen
*Improved Finished Print Screen
*Meshpoint Editing Screen
and some things more. 

We also played around with the different aspect ratios of the new TFT screens.

During this time i learned that only monospace fonts in a certain size can be used and only bitmaps (so no alpha...), that memory is limited and that many elements on a single screen can cause performance issues. 

Let me add, that even I have a little bit of experience with interface design, I have absolutely no experience with coding such a thing. So I believe that it is absolutely possible that there are ways to improve the current interface. I just wanted to let you know that it could eventually be very complex because of the limitations

oldman4U, 2020-08-26:
Hi. 

Please close this ticket in case you do not need it anymore. Thank you

Estivio, 2020-08-26:
I am really sorry, I have currently not the time to concretise my ideas. If the developement is still active in a few months I will come at you again with something more substantial than now.

closed 2020-08-26 by Estivio

oldman4U, 2020-08-26:
Your input is always welcome 👍🏻