bigtreetech/BIGTREETECH-TouchScreenFirmware #935: [FR] IDEA: Iconpack 'The Round Miracle' - Update for SW v26.x @ 2020/08/07

Acenotass, 2020-08-07:
Big SW 26.x update @ 2020/08/07. Added 9 new icons, redrawn 4 old ones.

[image]

[TFT24 Iconpack 'The Round Miracle' by Acenotass for v26.x@2020.08.07.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5042090/TFT24.Iconpack.The.Round.Miracle.by.Acenotass.for.v26.x%402020.08.07.zip)
[TFT28 Iconpack 'The Round Miracle' by Acenotass for v26.x@2020.08.07.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5042091/TFT28.Iconpack.The.Round.Miracle.by.Acenotass.for.v26.x%402020.08.07.zip)
[TFT35 Iconpack 'The Round Miracle' by Acenotass for v26.x@2020.08.07.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5042092/TFT35.Iconpack.The.Round.Miracle.by.Acenotass.for.v26.x%402020.08.07.zip)
[TFT43 Iconpack 'The Round Miracle' by Acenotass for v26.x@2020.08.07.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5042093/TFT43.Iconpack.The.Round.Miracle.by.Acenotass.for.v26.x%402020.08.07.zip)
[TFT50 Iconpack 'The Round Miracle' by Acenotass for v26.x@2020.08.07.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5042094/TFT50.Iconpack.The.Round.Miracle.by.Acenotass.for.v26.x%402020.08.07.zip)
[TFT70 Iconpack 'The Round Miracle' by Acenotass for v26.x@2020.08.07.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5042095/TFT70.Iconpack.The.Round.Miracle.by.Acenotass.for.v26.x%402020.08.07.zip)

Mistigris18, 2020-08-07:
tu est très réactif 👍

DrumClock, 2020-08-13:
HI,
again some new icons have been added.
Please update, Thanks

[image]

Mistigris18, 2020-08-14:
Regarde bien, ils y sont 😉

oldman4U, 2020-10-27:
@Acenotass 

Are you going to add the missing extruder tune icon?

el-quique, 2020-10-30:
Nice job. The only change if you can make is: Extrude_100.bmp to Tune_Extruder.bmp
Thx

Acenotass, 2020-10-31:
@oldman4U  Done!

oldman4U, 2020-11-02:
Hi.

Could you please check #1187 to see if this would be something you would also like do with your theme.

Thank you

Acenotass, 2020-11-02:
hi! 
@oldman4U 
If you want you can include it in the firmware, maybe it makes sense to make a mechanism for easier changing of themes. Thanks.

oldman4U, 2020-11-02:
I will mention you as the one who provided the theme. Ok!?

Maybe I can not do it, because the whole package becomes too big...

Will see.

oldman4U, 2020-11-02:
Is this OK?

[image]

Acenotass, 2020-11-03:
@oldman4U ok

cimoalpacino, 2020-11-09:
Great work! Thank you for sharing your design 👍

Mistigris18, 2020-11-10:
Très bon travail , merci je les utilises sur toutes mes imprimantes 👍👍

oldman4U, 2020-11-12:
Hi.

Somehow the Case light and chamber icons of the TFT35 folder seem to be broken. Could you please check.

Thank you

oldman4U, 2020-11-12:
And some more, like Apply Dec, Folder, Inc, Level Edge Distance, Machine Settings, Nozzle Up and Down,  Parameter, PID, Status Bed, Flow, Nozzle and Speed, Tune Extruder and Tuning.

oldman4U, 2020-11-12:
All others seem to be OK.

Acenotass, 2020-11-15:
HI @oldman4U 
I downloaded the archive from the post for TFT35, found no problems. What is your problem?

oldman4U, 2020-11-15:
Look at the attached screenshot.

Previous packs are fine. Only the newly added items seem to have the issue.
[image]

Acenotass, 2020-11-15:
re-download archive TFT35

oldman4U, 2020-11-15:
Perfect!!

Thank you

[image]

oldman4U, 2020-11-15:
I had to start a new PR, #1251 is now used. Please check if everything is fine.

Thank you

Acenotass, 2020-11-18:
@oldman4U 
what do you think of this icon?
[image]

oldman4U, 2020-11-18:
I would keep it more simple. The icons are already hard to understand
especially on small tfts.

oldman4U, 2020-11-27:
Hi. If you want you can close this ticket now. 

Thank you

closed 2020-11-27 by Acenotass