bigtreetech/BIGTREETECH-TouchScreenFirmware #943: [BUG] babystep in touchscreen mode

BsCmOD, 2020-08-08:
Hi, I want to report a just discovered issue. Since I compiled and update (the day before yesterday) my TFT35V3 (connected to a SKR1.4 turbo with the very last Marlin bugfix version) to 26.x version, babystep function it's no available during printing in touchscreen mode with the old version (V26) it was available also during this procedure. I've you got any idea to how to solve this issue? Just another thing, BTT had annunciated the 27 firmware version for their screens some time ago (about 2 o 3 weeks ago) but it was never released, the question is why? Thank'you for support

radek8, 2020-08-08:
Will be fixed #940

BsCmOD, 2020-08-08:
> 
> 
> > babystep function it's no available during printing in touchscreen mode
> 
> Will be fixed #940

Thank's man, any prevision?

digant73, 2020-08-09:
@BsCmOD the M117 popup message was removed (currently is simply commented out in the code) because too much verbose in many cases (e.g. it was displayed at each layer change). However, the message is displayed in the status bar

BsCmOD, 2020-08-09:
> @BsCmOD the M117 popup message was removed (currently it is simply commented out in the code) because too much verbose in many cases (e.g. it was displayed at each layer change). However, the message is displayed in the status bar

OH, ok, in fact in many cases was noisy anyway if it's used in smart mode it will visualized just for a few seconds a time. In status bar I can't see anything just the file name. Is there any options to enabled into menu extra?

digant73, 2020-08-09:
@BsCmOD probably in a next PR the M117 smart notification will be displayed.
The problem is that what is good for someone is not good for others

BsCmOD, 2020-08-09:
> @BsCmOD probably in a next PR the M117 smart notification will be displayed.
> The problem is that what is good for someone is not good for others

Yes, absolutely, for that function it could be possible add another options into the menu to turn on or off

PR #940 (2020-08-10): Addition of the BabyStep button
Addition of the BabyStep button to the More menu.
If a file generated with Cura plagin is printed to create an icon, the Baby Step button is missing in the PrintingMenu.
resolves #943 

Added BabyStep button to More menu.

Added BabyStep button to PrintingMenu, even when a model preview is displayed. Click on the model to display BabyStep

[image]

[image]

[image]
closed 2020-08-10 by bigtreetech