bigtreetech/BIGTREETECH-TouchScreenFirmware #944: [BUG] freezes on pre heat and printing menus when powering on with SD card inserted + runout filament error even when disabled

digant73, 2020-08-09:
A lot of freezes are now present since the last 3 weeks of updates.
They seems present every time the printer is powered on with the SD card inserted in the TFT slot.
Below a list of thypical freezes:

1) freeze when entering on preheat menu. All icons are displayed but only the text "ABS" in the first temp. icon is present. All the other texts such as PLA etc... and temp values are not dispalyed.
2) freeze navigating on file list before a print. That freeze seems present since the PR that added a thumbnail preview of gcode file.
3) often a filament runout error is displayed (althougth disabled on feature settings menu) during a print or when pressing on the print stop button.

All of this problems were not present in the past (I would say they are present since the last 4 weeks of updates)

I also verified that most of the freezes are due to the presence of the SD card when the printer is powered on. In order to avoid the freeze it seems that I need to remove the SD card before powering on the printer. Furthermore, a simple restart (on printers having the reset button) is not enough to avoid the freeze (even if the SD card is not inserted). The SD card must be removed and the printer must be powed off and the on

### Steps to reproduce

1. [First Step]
power on the printer with an SD card inserted on TFT slot
2. [Second Step]
go to preheat menu or go to print menu 
3. [and so on...]

**Expected behavior**
No freeze and no runout filament message must be present

**Actual behavior**
constant freezes are present, typically when printer is powered on with an SD card inserted in the TFT slot. I used different SD cards and the problems persisted. The problem was not present at all in the past (at least until 4 weeks ago everything was working fine)

oldman4U, 2020-08-09:
Have you updated the TFT folder, the firmware AND the config.ini file?

Sounds like a mismatch

digant73, 2020-08-09:
yes, I always install from scratch

closed 2020-08-16 by digant73