bigtreetech/BIGTREETECH-TouchScreenFirmware #947: [FR] GCODE VIEWER

jedisintraining3, 2020-08-10:
Having the ability to preview the look of the gcodes on screen would be super usefull when starting a print. 

I would like to be able to click the gcode and it would generate a picture of the item based on gcode before starting the print.

I have seen a lot of other full color screens do this.

oldman4U, 2020-08-10:
Hi. Using Cura as the slicer this is possible today. https://github.com/bigtreetech/Bigtree3DPlugin

RaMcHiPTech, 2020-08-10:
Any way to do this in simplify3d?

oldman4U, 2020-08-10:
This is Cura only as far as I know.

jedisintraining3, 2020-08-11:
Awesome Thanks!

closed 2020-08-11 by jedisintraining3

oldman4U, 2020-08-11:
Thx for closing the ticket

👍🏻

RaMcHiPTech, 2020-08-11:
> […](#)

Anyway to do it from command line on already existing gcode file?  If so could probably post process.