bigtreetech/BIGTREETECH-TouchScreenFirmware #949: [FR] Temperature Fluctuation Graphs

jedisintraining3, 2020-08-10:
Can you add temperature Fluctuation Graphs to visualize current, past, and required temperatures. I'm thinking of the graphs that are found in octoprint. I think I would add a lot of use and feature to the display.

oldman4U, 2020-08-10:
So far all features are available independent on the size of the Screen. Not sure how this works on smaller screens. Why not using Octo or Pronter for this?

jedisintraining3, 2020-08-11:
So, this is available for the bigger ones? I dont use octo because It would get rid of power det and filament runout functionality. Pronter is only USB.

oldman4U, 2020-08-11:
No. All sizes of TFTs are using the same interface.

You could get it with the WIFI module also. Which screen and mainboard are
u using?

oldman4U, 2020-09-03:
Hi.

Please close this ticket

jedisintraining3, 2020-09-03:
May the force be with you too, 

Always.

closed 2020-09-03 by jedisintraining3