bigtreetech/BIGTREETECH-TouchScreenFirmware #954: [Q] I am having issues updating to version 26.

MrJabu, 2020-08-10:
i have a BTT TFT35 and cannot get the new version installed.

i have grabbed the TFT35 folder and BIGTREE_TFT35_V3.0.26.x.bin file on an sdcard. when i boot it up it looks like it is updating but then jumps to a calibration screen with unreadable text. 

what am i doing wrong?

radek8, 2020-08-10:
TFT35 ? what version?

radek8, 2020-08-10:
Unzip to SD card and restart the display.
if it requires calibration, perform calibration
[TFT35V30.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5052950/TFT35V30.zip)

MrJabu, 2020-08-10:
> Unzip to SD card and restart the display.
> if it requires calibration, perform calibration
> Edit the INI file according to your own configuration
> 
> [TFT35V30.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5052950/TFT35V30.zip)

Ok that worked! 

why did your file work and the one on the github did not?

MrJabu, 2020-08-10:
> Unzip to SD card and restart the display.
> if it requires calibration, perform calibration
> Edit the INI file according to your own configuration
> 
> [TFT35V30.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5052950/TFT35V30.zip)

Actually, now my Marlin mode doesnt work.

carl1961, 2020-08-10:
that zip looks like stock, maybe try the unified

carl1961, 2020-08-11:
@radek8 where that zip come from? bin file different size.

radek8, 2020-08-11:
> 
> Dobře, že to fungovalo!
> 
> proč váš soubor fungoval a ten na githubu ne?

I have compiled a new version for your display.
There are obsolete bin files on github, but current INI files.
Together they may not work well ...

radek8, 2020-08-11:
> Like @radek8 asked what version TFT 3.5 ? theres many versions

Yes, there are many HW versions of the TFT35

radek8, 2020-08-11:
> @radek8 where that zip come from? BIGTREE_TFT35_V3.0.26.x.bin file different size (larger).

It is the latest compilation. Bin file is 29 days old :-(

carl1961, 2020-08-11:
Thanks @radek8  Good to know that, I did not notice that.

MrJabu, 2020-08-12:
> > Ok that worked!
> > why did your file work and the one on the github did not?
> 
> I have compiled a new version for your display.
> There are obsolete bin files on github, but current INI files.
> Together they may not work well ...

Can you post the fumes you used to compile the .bin

carl1961, 2020-08-12:
MrJabu You use vscode  https://code.visualstudio.com/, it's all documented on the BTT github   https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

MrJabu, 2020-08-13:
> @MrJabu You use vscode https://code.visualstudio.com/, it's all documented on the BTT github https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware Click on View full instructions towards the bottom Under Firmware

i actually did try to compile it before i posted the question here. i was able to create the .bin and did have the correct screen selected but it still didn't work

oldman4U, 2020-09-03:
Hello @MrJabu 

Seems this issue is solved.

Please close this ticket

Thank you

oldman4U, 2020-09-11:
@MrJabu 

Close this ticket or reply

closed 2020-10-09 by MrJabu