bigtreetech/BIGTREETECH-TouchScreenFirmware #955: after flashing newly custom built firmware (x.x.26), screen info displays "echo: M603 L0.00 U100".

jnle7, 2020-08-10:
After flashing newly custom built firmware (x.x.26) (for TFT35 E3 V3.0)  (Main board is SKR Mini E3 v1.2)
- turn on power, 
- wait for printer to connect
- screen info then displays "echo: M603 L0.00 U100".  

How do I stop this echo from showing?

jnle7, 2020-08-10:
Note: only happens in touch mode.    
Issue not happening in LCD12864 simulator mode.

jnle7, 2020-08-10:
re-opened - closed by mistake

carl1961, 2020-08-10:
check your filament sensor in  in touch mode (sensor needs to be on TFT for  in touch mode)

The M603 command configures automatic filament change parameters.

Notes
Requires ADVANCED_PAUSE_FEATURE.

jnle7, 2020-08-10:
I don't have any filament sensor connected to the TFT35 E3 and I did NOT modify any filament sensor settings in the config.ini:

#--------------------------------------------------------------------
# Filament Runout Settings (if connected to TFT Controller)
#--------------------------------------------------------------------

#### Default FIlament Sensor
# Options: [NONE: 0, Normal: 1, SMART: 2]
fil_runout:0

#### Filament runout inverting - invert the logic of the sensor.
# Options: [true: 1, false: 0]
fil_runout_inverting:1

#### filament noise threshold  - Pause print when filament runout is detected for this time period in (ms).
# time duration in ms
fil_noise_threshold:100

#### Smart filament runout detection
# For use with an encoder disc that toggles runout pin as filament moves
fil_runout_distance:7

carl1961, 2020-08-11:
do you have filament sensor enabled in marlin firmware?

jnle7, 2020-08-11:
No, the filament sensor is NOT enabled in marlin firmware (I double-checked configuration.h & _adv.h and they didn't have filament sensor enabled). 

So after I reloaded the BTT/github's pre-built firmware for TFT35 E3 v3.0 [29 days old] then the TFT35 boots up with Info screen = Ready (as expected).  

If I reloaded theTFT35 E3 firmware that I built using downloaded existing TFT touchscreen source currently on BTT/github then the error comes back ... so I'm sure the problem is with the current BTT/github TFT code ... I also checked the config.ini and all looks correct ... maybe this is a BUG that no one reports yet ...

Note: even while the Info screen displays "Echo M603 message"  everything still works as expected, printing completed ok ...

jnle7, 2020-08-11:
Issue seems to resolve itself after downloading the latest 10Aug2020 Master branch and recompiling code ... the "echo M603..." warning message is no longer displayed ... closing this ticket ... thanks!!

closed 2020-08-11 by jnle7