bigtreetech/BIGTREETECH-TouchScreenFirmware #956: Emergency Stop M112 Delayed

matbhe, 2020-08-10:
### Description
The emergency stop command M112 (em stop) will not be sent immediately after aborting a job with much long running g commands (marlin buffer full or something like this).

### Steps to reproduce
1. Create job like this
    G1 X1000 F100
    G1 X0 F100
    ... repeat lines above 1000 times...
2. Start job
3. Abort job
4. Push "EM Button / M112"

**Expected behavior**
Printer should stop immediately.

**Actual behavior**
Printer waits until buffer has space or someting like this and then sends M112. The delay depends of the duration of the g commands.

### Hardware Variant
TFT24 v1.1

### TFT Firmware Version & Main Board Firmware
TFT - Master
SKR 1.4 Turbo - marlin 2.0.x-bugfix

### Additional Information
When aborting short jobs or jobs which fast g codes then the delay is almost 0...

Msq001, 2020-08-11:
Hello, #960s should fixed this problem

matbhe, 2020-08-11:
The original problem was fixed with #960 - But #959 introduces another problem --> "no printer attached"....

Msq001, 2020-08-19:
Hello, can you confirm the lastest version?  Do you have any special settings for your Marlin temperature section? 
Because the "T:" keyword also appears in the response information of M115, So I added the "@" character to distinguish the response of M105 and M115

closed 2020-08-20 by bigtreetech

matbhe, 2020-08-20:
Hello. I'm using this display for a marlin based CNC machine (no extruders / temperature). Many of your customers like to do that. I think it is not the best idea to rely on temperature for that. This breaks the support for CNC machines. It would be great if you could add a feature toggle for cnc only support (0 extruders / no heat / no m221 command). Regards