bigtreetech/BIGTREETECH-TouchScreenFirmware #957: [FR] Probing point values

Guilouz, 2020-08-10:
Do you think is it possible to add values from BlTouch probing point on the screen like Creality printers ?

[image]

This is very useful to see if the leveling is correct.

oldman4U, 2020-08-11:
Hi.

Isn't this more or less exactly like this pinned ticket: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/288

radek8, 2020-08-11:
Look at this # 963
You cannot edit individual points, but you can check the point values.

Guilouz, 2020-08-11:
> #963

Yes that what I want, just see values. Is it not possible to have values in real time ? Like creality printers when it probing a point it show value directly.
Maybe create a new page when we start a leveling who show values, no ?

Hyrules, 2020-08-24:
I would like this as well. Coming from at CR10SPRO touchscreen it would be nice to see the values of the bed level. I can cope meanwhile with the Bedlevel visualizer in octoprint but it would make a nice addition to the TFT screen.

oldman4U, 2020-08-25:
@Guilouz 
@Hyrules 

Would it be possible for you to design a interface which works from 2.4 to 7“ display and describe the features? Only the functionality not the look.

Hyrules, 2020-08-25:
It really depends on the number of point you have to probe. Lets say someone has 4 point X and Y. That 16 point to show on screen. That should not be a problem for at TFT3.5 and bigger but let's say someone has decided to compile marlin with 8 point X and Y that 64 points on screen. I don't believe there is enough room on the 3.5 and smaller to display those 64 points.

oldman4U, 2020-08-25:
I agree.

So this function should not be available for 2.4“ screens?

Or what would be the solution for you?

Hyrules, 2020-08-25:
This should be really dependant of 2 things the TFT size and the number of points. If someone has a TFT3.5 but decide to use 64 points it might not show properly on screen too. It might be available but only list the points on screen as a XY : Value pair one under the other.

oldman4U, 2020-08-25:
Take some pieces of paper and a pen and make the design. This could help a
lot.

Hyrules, 2020-08-25:
I'll try to get a design for you asap.

oldman4U, 2020-08-25:
No hurry.

So 2 designs. One where the grid fits and one where it does not.

I would also need the design how the grid is then edited.

radek8, 2020-08-25:
@oldman4U Traffic-light is already working on it :-)
See #963

oldman4U, 2020-08-26:
Hi radek8.

I hop traffic light is as good in UI design like in coding😁

Many others including myself have been working on it but for good reasons nobody was able to release it so far. My experience is that it is almost impossible to get a feature rich solution which works fine on all sizes of displays. So for me every additional input is welcome and as soon as traffic light adds letters to his grid, the problem will start. With the current monospace font and the limited amount of pixels available there is simply not enough space to make it happen. Do you know that the TFT43 and 50 have less pixels than the TFT35. So it is a question of the size of the screen, the number of pixels available and the aspect ratio. 

This is a UI developers nightmare

Hyrules, 2020-08-26:
Basically I'm not good either at making ui. I'm hopefully a good coder but not a gui maker. I maintained an app for the best part of the last 6 -7 years and I always had troubles designing a gui.

oldman4U, 2020-08-26:
Also good coders are always welcome😊

Hyrules, 2020-08-26:
I might try myself at forking this repo and make some bugfix and add features. I would just need an extra TFT because the only one I have is on my printer and I don't want to mess around with it too much.

oldman4U, 2020-08-26:
Welcome to the "I have two printers - one is printing and one is just parts" world;-)

oldman4U, 2020-09-11:
Hi.

What is the status of this ticket?

oldman4U, 2020-10-11:
Hi.

I am just working on a new ticket which collects all the features which have been requested in the past. Yours have been requested before and after, so could you please close it.

Thank you

closed 2020-10-11 by Guilouz