bigtreetech/BIGTREETECH-TouchScreenFirmware #958: WIFI REPETIER support

Virtualchronos, 2020-08-11:
Hello !

I've baught a TFT 7.0 and there's plenty of incompatibility problems with Repetier :

Bltouch commands are not the same so it doesn't work
When there's "checksum errors" you must restart the screen to re-enable communication"
But the most important problem is ESP3D : I can't anymore use it since most of the commands doesn't work. The scren seems to filter the web commands so I can't anymore connect the ESP to repetier host. For the web interface, the only working commands seems to be movement and heat management. I can't remotely launch a print and that's a huge problem for me

I tried to put the esp on UART1 and THT7.0 on UART3 but the only result is endless beeps with "checksum error" message as soon as rx-tx on UART1 is connected

oldman4U, 2020-10-20:
Hello Virtualchronos.

Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

Virtualchronos, 2020-10-21:
Can you tell me what does it solve in my request please ?

Télécharger Outlook pour Android<https://aka.ms/ghei36>

oldman4U, 2020-10-21:
It does not solve the issue, but it helps to bring your feature request to a more prominent place than an unanswered ticket in the middle of page 2.

Virtualchronos, 2020-10-21:
I don't understand.  If my issue is not resolved, how to mark it resolved could help to résolve it ?

Télécharger Outlook pour Android<https://aka.ms/ghei36>

oldman4U, 2020-10-21:
Because in exchange of closing it, it will be part of the Feature Request List on the Main Page and therefore it will be easier to find it.

If you are not comfortable with closing it, you can keep it opened and the ticket will be unchanged. In this case I will remove it from the list, to avoid to have the same request twice.

Virtualchronos, 2020-10-21:
Ok now I understand

Télécharger Outlook pour Android<https://aka.ms/ghei36>

closed 2020-12-27 by stale[bot]