bigtreetech/BIGTREETECH-TouchScreenFirmware #962: Change mode with the rotary knob

iBlack22, 2020-08-11:
*I would like to switch between marlin and touch mode with the rotary knob ...... 
I press the rotary knob for 3 seconds then the menu appears on marlin or touch mode, and the selection which mode i want to use i would like to make with the rotary knob and not press the display. 

Does anyone have an idea what I have to activate in the FW? earlier this was possible .....

radek8, 2020-08-11:
It's not a good idea,
compressing the encoder has side effects

bigtreetech, 2020-08-12:
The latest version of the firmware is restored, But the precompiled firmware has not been updated, so you can compile the latest. .bin yourself.

> It's not a good idea,
> compressing the encoder has side effects

As mentioned above, If the cable of EXP-2 or EXP-3 port is connected, the rotary knob will bring side effects, 
When operate rotary knob, At the same time, Marlin will recognize the actions of the knob to execute the corresponding commands, but the screen is not in Marlin mode, so it is impossible to see what happened in Marlin in real time, so unexpected things may occur.

traffic-light, 2020-08-12:
@iBlack22 This one can be closed. It works.

closed 2020-08-13 by bigtreetech