bigtreetech/BIGTREETECH-TouchScreenFirmware #964: BTT SKR Mini E3 V1.2 and BTT TFT35 E3 V3.0 on Ender 3 Pro with connectivity issues on TFT’s RS232 and SWD ports with No Printer Attached.

Ve6hgw, 2020-08-11:
I upgraded the SKR Mini E3 V1.2 board and TFT35E3 V3.0  display at the same time and everything worked perfectly for a week then  the touch screen went black.  I checked the connections on the board and display and everything seemed okay but the screen remained black.

After much testing I discovered both the Marlin and touch screen modes light up if I leave the ribbon cable on EXT3 and connected the serial cable to the SWD port on  the TFT35.  I can print and everything works fine on the Marlin screen but I can’t clear the No Printer Attached message on the touch screen by toggling the UART settings to 115200.  Therefore I can’t print or send gcodes from the touch screen.

With the proper configuration of the ribbon cable attached to EXP3 and the serial cable attached to RS232 the TFT35 is black.  If I then remove the serial cable from RS232 the TFT35 remains black.  If I then connect the serial cable to the TFT35’s  SWD port I have access to both screen modes and I can print from the Marlin screen but I can’t clear the No Printer Attached on the touch screen.  I always power off before changing any connections and I haven’t changed any of the connections on the motherboard since the original working upgrade.  I haven’t made any firmware changes or printer mods other than install a plug and play filament runout sensor to the TFT35  that worked perfectly.

Back to the ribbon cable on EXT 3 and the serial cable connected to the TFT35 SWD port with Marlin mode working okay and No Printer Attached on the touch screen mode, here’s what I’ve tried from the Marlin screen:
Flashed the original firmware  TFT35 E3 V3.0.26.x to the TFT , flashed Bugfix 2.0.x to the motherboard from its original microSD card.  Both BIN files changed okay to CUR.  Flashed rhe latest Bugfix  2.0.x from Github, Initialized EEPROM, restore failsafe and saved EEPROM.  I also replaced the ribbon and serial cables and tried V3.0.25.x  on the TFT35 but no change.  If I remove the serial cable and attach the ribbon cable to EXT1 I have the touch screen with No Printer Attached but no access to the Marlin screen.  Strange.   I haven’t done any compiling yet therefore I haven’t been able to check the firmware’s port or baud rate settings on the SKR Mini E3.

This upgrade is awesome and I miss the touch screen’s additional functionality.   Is this a firmware issue or hardware issue?  What do I need to do to get my SKR Mini E3 and TFT35 E3 working properly again?
Thanks,
Drew.

bigtreetech, 2020-08-12:
Hello, RS232 interface should be used. SWD interface is totally wrong.
` serial cable attached to RS232 the TFT35 is black. If I then remove the serial cable from RS232 the TFT35 remains black`
It seems that the line sequence is reversed, serial cable is 5pin (4Pin + 1pin), and the 1 pin should be connected to the Reset side in SKR-mini-E3-v1.2.
Can you attach the wiring diagram of your screen and motherboard？

oldman4U, 2020-08-12:
This worked fine but suddenly stopped working. Sounds like a hardware issue to me.

Ve6hgw, 2020-08-12:
Connected the serial cable to RS232 and powered up.  TFT35 still black.  Moved the serial cable back to SWD and I have a functioning Marlin screen and No Printer Attached on the touch screen.

If it's a hardware issue how do I determine if it's the display or motherboard?

[image]
[image]

radek8, 2020-08-12:
The SWD connector has a 3.3V power supply, RS2232 is at 5V.
Couldn't the damage have occurred?

[image]

radek8, 2020-08-12:
Marlin mod works and touch doesn't it?
Do you have the correct TFT serial port set in Marlin?

Ve6hgw, 2020-08-13:
I agree that damage could have occurred when connecting the serial cable to the SWD port but I didn't try that till after the TFT35 stopped working.  I'd like to determine if the problem is hardware and if so is it the display or motherboard.    My gut feel is I have a motherboard issue rather than a firmware issue.  I haven't learnt Visual Studio Code and PlatformIO yet so I don't know the serial port assignment but I guess the time has come.   I'm anxious for feedback on the wiring in my photos.
Thanks,
Drew.

oldman4U, 2020-08-13:
I believe it was broken before you connected to the SWD port. 

For me the fotos show the correct cabling. 

Besides replacing one part there is no way to know which one is the broken one. But i would guess it is the TFT.

radek8, 2020-08-13:
You write that you flashed the new Marlin
You need to know which port you used as serial for the TFT display.
Otherwise it is a lottery bet ...

radek8, 2020-08-13:
Does the display work in Marlin mode and does the printer print?
In touch mode, the display shows that the printer is not connected?

Ve6hgw, 2020-08-13:
Yes the Marlin display works fine connected to SWD and I can print.  The touch mode display lights up and I can move around the menus but I can’t clear the No Printer Attached message therefore can’t communicate with the printer from the touch screen.

Looks like I need to learn VS Code and check out the serial port configuration before replacing hardware.

radek8, 2020-08-13:
> Yes the Marlin display works fine connected to SWD and I can print.  The touch mode display lights up and I can move around the menus but I can’t clear the 

No SWD, but EXP3

oldman4U, 2020-08-13:
As soon as the display gets power from EXP OR Serial port it powers up and both user interfaces are available completely independent of the other components and their settings. If the TFT then can connect to the mainboard properly or not and therefore if you can print with a mode then depends on the mainboard and the proper configuration. So in your case this means that the EXP based connection for Marlin mode works, but that the serial port of the TFT can not „talk“ with the mainboard. This connection requires the cable to be connected to the RS-232 connector. As long this is not the case the TFT can not connect to the printer.

Ve6hgw, 2020-08-13:
No SWD, No RS232, Yes EXP3 - black TFT.
No SWD, Yes RS232, Yes EXP3 - black screen.
No SWD, Yes RS232, Yes EXP1 - black screen.
Yes SWD, No RS232, Yes EXP1 - two lines on TFT at all times.
Yes SWD, No RS232, Yes EXP3 - Marlin prints okay but touch screen is No Printer Attached.
No SWD, No RS232, Yes EXP1 -  touch screen menus work but No Printer Attached and no printing but the Marlin screen displays LCD12864 but I can't access any of the Marlin menus.

If the touch screen needs the RS232 port to talk to the printer but the TFT is always black if I connect to the RS232 port does it make sense that its a hardware issue rather than a serial port assignment in the firmware?  In other words, in a perfectly working system if you just change the serial port assignment in the firmware to a wrong port, will the TFT light up in touch screen mode and display No Printer Attached?

radek8, 2020-08-13:
Only connect RS232. Will touch mode be available?
Maybe even with "No Printer Attached"

radek8, 2020-08-13:
You did not use an external 5V power supply module.
Do you happen to have a badly connected orange jumper?

[image]

oldman4U, 2020-08-13:
You had a properly working system when you connected the ribbon cable to EXP3 and the other cable to RS-232 - and this is the correct setup. When you connect the RS-232 cable now, you get a black screen - so the hardware is broken. Point. Which hardware we do not know, could be mainboard, cable or TFT.

Like i wrote before - as soon as the display gets power, Marlin Mode and Touchscreen mode are available, independent of the mainboard. I never tried it out, but i guess attaching a 5V power and ground to the TFT will also give you a "full working screen", even you can not do anything with it.

To answer your question:
_In other words, in a perfectly working system if you just change the serial port assignment in the firmware to a wrong port, will the TFT light up in touch screen mode and display No Printer Attached?_

Yes. The display will light up as soon as it gets power.
In case there is no connection because the mainboard, the cable or the TFT is broken or the serial port or the baud rate is wrong or does not match, you will get "No Printer Attached" because the TFT can not find the mainboard.

Connecting the RS-232 cable prevents your display from booting correctly, which makes me believe that the problem is hardware related.

And again. Connecting the EXP OR the RS-232 cable alone makes the TFT light up and shows the two modes.
RS-232 connection is needed for touch screen.
EXP is needed for Marlin Mode.

On one of my printers i removed the encoder wheel and the EXP ribbon cable and use only the Touch Screen function and the RS-232 cable. Works perfect.

radek8, 2020-08-13:
I think there is a problem with the power supply of the TFT display
Connect the display as shown in your photo and also connect the 5V connector on the TFT with the 5V on the Marlin board.
Check the serial port speed 115200 or 250000 according to the vMarlin setting. Then both modes will work if the correct port is set in Marlin

oldman4U, 2020-08-13:
And maybe replace the self made RS-232 cable and try the one which came with the display for a test.

Ve6hgw, 2020-08-14:
Using an external 5 volt power supply to the TFT, NOT FROM THE MOTHERBOARD.  Is no common ground an issue with this test?

External power to TFT35 with no other connections -  Marlin screen displays LCD12864 Simulator and no other menus, touch screen menus all available, no printer attached.
External power to TFT and only RS232 cable attached and printer powered up - TFT LED on but screen black.
External power to TFT and only ribbon cable to EXP3 and printer powered up - TFT LED on but screens black.
External power to TFT and ribbon cable to EXP3 and RS232 cable attached and printer powered up - TFT LED is on but screen is black.  This is the proper configuration.
External power to TFT and ribbon cable to EXP1 and printer powered cup - Marlin screen displays LCD12864 Simulator and no other menus, touch screen menus all available, no printer attached.
External power to TFT and ribbon cable to EXP1 and RS232 cable attached and printer powered up -TFT LED is on but screen is black.

Tried a different serial cable, same results as above.  I still haven't confirmed serial port config in firmware but I think the TFT is probably okay and there's a problem on the motherboard with RS232 circuit.  What do you think?

oldman4U, 2020-08-14:
This all makes sense.

But i believe the TFT is broken. Maybe i find time to try the external 5v
to the TFT today.

radek8, 2020-08-14:
> External power to TFT and only RS232 cable attached and printer powered up - TFT LED on but screen black.

Only connect the RS232 cable, but do not connect the RST landscape pin.
Will touch mode then be available? What happens if you then connect the RST pin?

Ve6hgw, 2020-08-14:
Only RS232 cable attached, RST pin not attached.  Powered up and Marlin screen displays LCD12864 Simulator and no other menus available, touch screen menus available with No Printer Attached.  Toggled UART speed and no printer attached remained.

Powered down, attached RST pin, powered up, TFT LED on, screen black, no Marlin or touch screen menus.
Powered down, removed RST pin, powered up, Marlin screen displays LCD12864 Simulator and no other menus available, touch screen menus available with No Printer Attached.  Toggled UART speed and no printer attached remained.

radek8, 2020-08-14:
Great. So we already know that there is a short circuit on the RST pin.
Do not connect RST anymore.
leave RS232 connected (RSTtin limit) and connect EXT3.
you should have a marlin available for touch mode. It is so?

radek8, 2020-08-14:
My guess
If you have connected an RS232 cable to the SWD connector
You connected 5V to the RST pin.
This is 3.3V logic, this pin may have been destroyed and the printer board will constantly start the display if the RST pin is connected.

Ve6hgw, 2020-08-14:
Serial cable attached to RS232 and RST pin not connected to motherboard..  Ribbon cable connected to EXKP3.  powered up andTFT LED is on but no Marlin display and no touch screen display. Move ribbon cable to EXP1 and Marlin has LCD12864Simulator only and touch screen menus work but can't clear no printer attached.

radek8, 2020-08-14:
OK
You have to break the wire that leads to the RST pin on the flat wire.
Then connect the cable to the EXP3 connector
The Marlin simulator should work and a touch menu should be available.
You can do it?

[image]

radek8, 2020-08-14:
Ještě než přerušíš vodič, Můžež ke kabelu EXP3 připojit původní displej z Enderu, zda bude dostupný Marlin na tomto displeji?

radek8, 2020-08-14:
> Move ribbon cable to EXP1 and Marlin has LCD12864Simulator only 

Query.
Does that mean you only see the sign or is there a Marlin status screen?

radek8, 2020-08-14:
You must have a Marlin with a display controller
# define CR10_STOCKDISPLAY
# define SERIAL_PORT 2
# define SERIAL_PORT_2 - 1

if you have
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
there will be constant restarts, and the display will not display anything
Where did the Marlin you recorded go to?

radek8, 2020-08-14:
If you can't compile Marlin yourself, download the precompiled from here
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2
And flash the board with Marlin
RS232 connection without RST pin
Connect the EXP3 connector and it should work
In touch mode, set the speed to 115200.

Ve6hgw, 2020-08-15:
I’ll hold off on breaking the RST wire .  I’ll dig out the original Ender display and get back to you.  I’m in Canada, mountain daylight time, do you want me to phone you?

Answer to your query, on Marlin mode I only see the LCD Simulator message and don’t see the Marlin status screen and can’t access any Marlin menus.  

I’ll hold off on flashing your file for now.
Thanks,
Drew.

Ve6hgw, 2020-08-15:
SKR Mini E3 V1.2 connected to original Ender 3 display EXT3, power up and display is blue with a lighter blue border.  Rotating and pushing the control knob does nothing.

No problem to cut Reset wire on ribbon cable to EXT3 when you’re ready.

radek8, 2020-08-15:
From what you write, it looks like you have Marlin configured for the EXP1 connector, but on the SKR mini board you have the EXP3 connector.
Therefore, resets occur.
Flash the correct Marlin configuration. I won't move without it.

Ve6hgw, 2020-08-15:
Serial cable to RS232, RST not attached on motherboard, ribbon cable on EXP3.  Powered up and TFT LED is on but screen is black.  Control knob movement does nothing.  Your new firmware did not load, bin file did not change to cur.

I removed the SD card and moved the ribbon cable to EXP1 and I can still get an active touch screen with no printer attached and only the LCD Simulator message on Marlin screen.  Should I try to flash your firmware with this wiring configuration?

radek8, 2020-08-15:
Can you connect to the printer from a PC via USB without a display connected?
Use the Pronterface program

Ve6hgw, 2020-08-16:
I managed to load Pronterface on my Mac okay but I can’t connect to the printer.  I get serial connection error when I try to connect  The Port button won’t display an active port.  I’ve tried many cable connection configs but no luck.  The closest I can get is connect to only EXP1 and the USB cable.  This will power up the display in touch screen mode with no printer attached message.  I can use the touch screen to Disconnect for computer control but  Connect on Pronterface gives the serial connection error.   If I just enter M503 the message says printer offline.
With no display attached, just the USB attached I can’t get Pronterface to connect to the printer.

radek8, 2020-08-16:
Can you connect to the printer from a PC via USB without a display connected?
Use the Pronterface program
If so, record a new Marlin without a display attached.

Ve6hgw, 2020-08-16:
Referring to my previous comment on my Pronterface tests it looks to me like it may be time for hardware replacement.  What do you think?  Is this the time to upgrade to the SKR Mini E3 V2.0?  If the display is bad I’d replace it with another TFT35 E3 V3.

radek8, 2020-08-16:
Your SKR miniE3 will probably be dead

Ve6hgw, 2020-08-16:
That’s enough incentive to upgrade to V2.0.  I’ll keep this open and let you know the outcome.  Thanks very much for your expertise, help and time spent.  Much appreciated.
Drew.

oldman4U, 2020-08-16:
Maybe you want to replace it with a board with external drivers. In case of a failure you get a new mainboard for around 15.- and if the driver breaks this is just 5.-

Ve6hgw, 2020-08-17:
I never thought of that.   I’ve been more of a plug and play type but you have my interest.   I’d appreciate your board and driver recommendation for my Ender 3 Pro.  Thanks.

radek8, 2020-08-17:
I also have Ender 3.
Upgraded BTT SKR1.4 + TMC2209 + BTT TFT35V3 E3
I printed my own case for electronics

oldman4U, 2020-08-17:
Ender3 Pro with standard case SKR E3 DIP with BTT TMC 2208 and TFT 35 E3 v3

radek8, 2020-08-17:
> Ender3 Pro with standard case SKR E3 DIP with BTT TMC 2208 and TFT 35 E3 v3

OK, but you only have 4 motor drivers.
You cannot upgrade to two-color printing

oldman4U, 2020-08-17:
Thats right. Not sure the new motor expansion could be added or not.

The DIP is a direct replacement of the Melzi which also has single
extruder.

And i do not use dual extrusion with my SKR pro based second printer. Too
complicated and too expansive when used for special support filament.

My 2 cent

radek8, 2020-08-17:
I agree with SKR E3 DIP, but together with TMC2209 :-)

oldman4U, 2020-08-17:
With sensorless homing😁

Works great on my second printer but now we are far away from plug and
play.

closed 2020-08-24 by Ve6hgw