bigtreetech/BIGTREETECH-TouchScreenFirmware #965: [FR] M600 support in Unified menus

jariivanoff, 2020-08-12:
**Is your feature request related to a problem? Please describe.**
I like to print with different colors, and change filament at a certain layerheight. I use to make that in slicer, and it sends an M600 command. Now that I have installed an TFT35 v2.0 I have no way to continue the print since there is no support for that in the menus and the TFT2.0 have no Marlin mode which I could switch over to and make the print continue after filament change.

**Describe the solution you'd like**
A button that I can press to make the print continue after a M600

**Describe alternatives you've considered**
Getting an other display with Marlin support or go back to my LCD12864

**Additional context**
Add any other context or screenshots about the feature request here.

bigtreetech, 2020-08-13:
Hello, It has supported M601 for gcode pause after #764 
And after #971 The default Settings included M600/M601
They are consistent with the action of manual pause(click pause button in touch screen)
Can this solve your needs?

ETE-Design, 2020-08-13:
@bigtreetech Is it possible to use this feature if my sensor is connected to the mainboard and not the screen? End where in firmware do I define the pin so it know It's connected to the mainboard?

jariivanoff, 2020-08-13:
Well.. the thing is that if you use a slicers feature of M600 the board won't know that the printer has paused.. if it  comes with the gcode so to say... and besides that.. I haven't found the pausfunction yet.. but that is perhaps since I always print from Octoprint and never use the display for pausing...

bigtreetech, 2020-08-14:
> Well.. the thing is that if you use a slicers feature of M600 the board won't know that the printer has paused.. if it comes with the gcode so to say... and besides that.. I haven't found the [...]

@jariivanoff Hello, Please come #966  to discuss it

bigtreetech, 2020-08-14:
> @bigtreetech Is it possible to use this feature if my sensor is connected to the mainboard and not the screen? End where in firmware do I define the pin so it know It's connected to the mainboard?

Hello, it also need support for HOST_ACTION_COMMANDS and HOST_PROMPT_SUPPORT in Marlin and Touch Screen, TouchScreen has not  implement it now.
 Please come #966 to discuss it

closed 2020-09-01 by bigtreetech