bigtreetech/BIGTREETECH-TouchScreenFirmware #966: [BUG] Pause does not park TFT50

Rebel3D-CZ, 2020-08-12:
The pause is not parked, or only sometimes at defined XY coordinates. Printing stops and retracts. After selecting "Resume", deretration is performed and printing continues. I use it for a filament sensor connected to a TFT50.

The ideal solution would be to implement M600 (ADVANCED PAUSE FEATURE), or add similar / same options to FW TFT https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/965

sorry for english, i use translator :)

bigtreetech, 2020-08-13:
Hello, It has supported M601 for gcode pause after #764 
And after #971 The default Settings included M600/M601
They are consistent with the action of manual pause(click pause button in touch screen)
Can this solve your needs?

Rebel3D-CZ, 2020-08-13:
582/5000
I don't understand it or it's working badly.

1. I compiled the latest version from github
2. in Marlin FW I have M600 enabled = verified from 1284 LCD. It stops, parks at the positions specified in the Marlin FW, ejects the thread and waits for insertion, and continues by pressing. Then it returns to the print position and continues. It's alright.
3. Press manual pause on the TFT50. Printing stops and nothing happens. After pressing the "Continue" button, printing continues. It is bad.
4. When I enter and send the M600 on the TFT50 LCD terminal, printing stops and nothing happens. This is also wrong, so the M600 does not send to the Marlin FW?

bigtreetech, 2020-08-14:
If the touch screen send M600 to Marlin for pause, Marlin must Enabled `EMERGENCY_PARSER` , `HOST_ACTION_COMMANDS` and `HOST_PROMPT_SUPPORT`, and the touch screen needs to send M876 to Marlin for resume, otherwise the touch screen will lose communication with Marlin, Because Marlin stops responding the normal gcode only responds M876 (when enabled `EMERGENCY_PARSER` )  in the M600 Park state.
And the analysis of the interaction of HOST_ACTION_COMMANDS has not been implemented now, So the current touch screen dont send M600 to Marlin, instead, touch screen will send manual pause command to marlin for pause, if the config.ini is seted in the figure below
[image]
the pause gcode is
```
<Set absolute coordinate mode>
G1 E<current E - 15> F600
G1 Z<current Z + 10> F6000
G1 X10 Y10 F6000
<Restore coordinate mode>
```
the resume gcode is
```
<Set absolute coordinate mode>
G1 X<X coordinate before paused> Y<Y coordinate before paused> F6000
G1 Z<Z coordinate before paused> F6000
G1 E<E coordinate before paused - 15 +16> F600
G92 E<E coordinate before paused>
<Restore coordinate mode>
```

so If this can't solve your needs, please wait we add support for `HOST_ACTION_COMMANDS` and `HOST_PROMPT_SUPPORT`

Rebel3D-CZ, 2020-08-14:
The problem is that the TFT does not send (or only sometimes) the nozzle parking commands set in config.ini during the pause. The nozzle therefore stops only in the last printing position and the retraction set in config.ini (pause_retract: **R10** P16) is checked.
I'll try to make a video later so we can understand each other.

ETE-Design, 2020-08-14:
@bigtreetech So when Will you implement the parts needed for "Real" M600 to functions with Marlin?

bigtreetech, 2020-08-17:
@ETE-Design   It's being implemented in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/984, but more testing is needed

closed 2020-09-01 by bigtreetech

Rebel3D-CZ, 2020-09-13:
In case of emergency, a pause in FW TFT50 is enough for me, but it does not work as it should. Skips parking to X / Y parking coordinates (pause_pos: X1 Y1), stops printing only. Performs only stop / resume and retraction parameters (pause_retract: R0 P1).