bigtreetech/BIGTREETECH-TouchScreenFirmware #967: Duplicate features and icons

rightawy, 2020-08-13:
### Description
Since last few PRs there are more duplicate features beeng added.
Like the baby stepping is in side both ABL and Manual Leveling, while it should have been in the move menu.
The EEPROM save restore options now available in 3 places: customenu, in parameter settings, and in the Machine settings where the user needs to juggle multiple times to select correct option.
There are two different icons for disabling steppers and save/load buttons but all of them do the same thing.
The new save and load button are both same floppy disk with just different back color.
In the z offset menu there are 4 toggle buttons, it will be easier to go back rather than clicking 4 times to disable steppers then 4 time again to get to the save button, this could have been done by asking the user to save or not when clicking on the back button.
The 100mm button in tuning menu is not useful as it does not check for heating and the same can be done through extrude menu which also allows to change length (between 1mm and 200mm) and select different speed. The tuning menu should be inside the machine settings.

traffic-light, 2020-08-13:
This is not really a bug.
But I agree with some points.

traffic-light, 2020-08-13:
Oh. By the way. I'm working on a new version for the extruder tuning.

digant73, 2020-08-13:
@rightawy 
The correct location for EEPROM save, reset ... should be only on MachineSettings menu. The correct place, easy to see, easy to use (big icons). Opening a small menu, scrolling at the end and then pressing on a small text could be a problem for a lot of people with handicap on their hands or eyes.
I see that in the last PR, by default that menu is now disabled. It seems that people doesn't have any problem here or prefer the most complicate approach.
The function for the steppers are different. One is to disable all steppers, while the other (used in other menu such as Leveling and Z Offset) disable only X and Y axis.
In the Z offset menu, that is the best layout to provide all those features according to the status of z offset process. Exiting from the menu will always disable z offset process, if not already stopped by the user. Asking to save on exit will not allow you to be notified again in case you denied to save the first time. The menus must be contextual. Now in the manual leveling menu, babystep is no more available while it allowed to fine tune the leveling as final step. Another bad choice.

traffic-light, 2020-08-13:
Currently working on the tune extruder.
You can find it here:

rightawy, 2020-08-14:
@digant73  You need click on screen 7 times to go to reset eeprom icons and 4 click to go to reset EEPROM in the custom screen. Why would anyone think that the M500 command will save only parameter settings? and anyone using a 3D printer knows that there is only one command to save the settings to EEPROM. even when using reset from marlin firmware menu it reset all of the settings including the ABL & UBL states. maybe change the name"Custom" to G-Codes" or "Macros" or to "Commands" to make it more sensible.

What is the point of handicap in this? The list view is not hard to use. The buttons in the Terminal keyboard are smaller than the list view. even the keyboards on mobile phones have much smaller lists than on these displays. These TFTs are resistive touchscreen which works best only with a stylus. I do not think it is a good idea to add a new icon for every new gcode shortcut.
I read earlier somewhere that the flash storage is almost full with new icons and with the last PR today, I can see in the info screen, it has filled up 94% already. 
Take the nozzle up-down icon, in place of them the z-axis up-down button could be easily used and mean the same thing also. 
The same way the EEPROM buttons and save-load buttons. The disable stepper button can use different text to tell their function.

In the z offset it is bad logic. Many printers have software end-stops always disabled for z-axis because of this turning it OFF automatically will result in malfunction and what if someone forgets to press the save button because the button is not visible? Why have the toggle buttons again when people were already complaining that it was hard to switch between more than 3 options.

traffic-light, 2020-08-14:
Macro's is indeed a better name for it.

And about the flash storage: Using one icon on multiple buttons should help.
And instead of making a complete picture for a button it's also possible to make the background in software instead.
This will be a lot of work but will save some memory

digant73, 2020-08-14:
@traffic-ligth
The use of resource could be drastically improved simply removing a lot of languages. I think everyone understand the meaning of words such as print, stop, start, temperature etc...
There should be clear rules to follow to submit a PR. E.g. the use of main memory could be drastically improved using local variables. But it's an open project...

@rightawy In fact in my PR I spent most of the time making cleaning, removing from all language files labels with the same value, move global variables to local variables and fixing bugs here and there etc.... The problem with flash memory will be solved simply removing at least 15 languages. You complained of babystep menu was present in the wrong menus but then you justify that it's better to use Custom menu for eeprom. The menus must be provided where it makes sense they are provided and must be also contextual (babystep is needed on both ABL and manual leveling but now you have to leave that menus to use babystep). Once a menu already exist (e.g. Babystep) it can be reused in other menus without the need of extra resources (local variables are used). About the usability of list menu, try to use a list menu on 28" TFT. Honesty I do not understand most of your critics even regarding Z offset and other menus. Maybe you didn't see their previous implementation or you ignore that the menus can normally contains only 8 sub menus. It's better to have the save button in the menu requiring that features (babystep, z offset etc...). Try to use the Parameter menu, make a change in some parameters, exit from the menu, don not accept to save, then change idea and you want to save the settings, move again in the menu, leave the menu and see if a prompt to save changes is displayed. There is simply no consistency in that feature

traffic-light, 2020-08-14:
@digant73 I agree with you. I see no problems with the current menu's. And those toggle buttons are great! It's easy to use and now it fits inside one menu.

rightawy, 2020-08-15:
@digant73 

> You complained of babystep menu was present in the wrong menus but then you justify that it's better to use Custom menu for eeprom. The menus must be provided where it makes sense they are provided [...]

let me be clear, my complaint was about baby-stepping showing in two different places when it should be at one place alongside the Leveling options not inside them because baby-stepping is an independent feature in marlin firmware which can be enabled even if ABL is completely disabled in marlin firmware. Why show baby-stepping as a Leveling only feature in the TFT? The same I am asking about EEPROM buttons if you want them separate then why you still showing them in other places where you do not want them?
 
> About the usability of list menu, try to use a list menu on 28" TFT. Honesty I do not understand most of your critics even regarding Z offset and other menus. Maybe you didn't see their previous implementation or you ignore that the menus can normally contains only 8 sub menus. 

And you are ok with the keyboard in the g-code terminal screen? Screen size is a personal choice. you should use one which you will be able to use comfortably. It will be a bad decision for a person who knows he has handicap on his hand to and still buys a smaller screen which will be hard to operate by him. No one will suggest him to buy the smaller screen. Even I have big fingers and I know what screen is good for me.

> It's better to have the save button in the menu requiring that features (babystep, z offset etc...). Try to use the Parameter menu, make a change in some parameters, exit from the menu, don not [...]

If I have to save the settings after having the Idea I will go back to the parameter menu and press save settings button. Why would I look for a save prompt when I already declined it in the first place?

> You complain of toggle buttons were used again when people already complained of that. Can you provide those links? I didn't read any of that in this repo.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/733#issuecomment-639316913 oh look you also commented on that PR.

digant73, 2020-08-15:
@rightawy 
But you opened a BUG report mentioning only personal preferences of you preferred layouts instead of opening a request for an improvement.
You didn't even consider that people could use the stock TFT provided in the printer (e.g. 28") instead of an external large TFT and that the use of list menus is the last chance when you have a lot of items to handle (e.g. Parameters menu).
#733 has nothing to do with toggle menus. Heat and pheheat menus were drastically changed by the PR author, and I perfectly agreed with most of the people to not drastically change something that was working fine.
The right points in all of your critics are to reduce some icons (stepper and reuse EEPROM icons also for "load" and "save" in ABL menu and to take into consideration that min. stop activation in ProbeOffset menu can produce side effects in some Marlin configurations (that you can avoid using Babystep menu).
But these are typical improvements and cleaning activities periodically made in the project (such as also the cleaning in all the language files to remove duplicates or to provide general words that could be reused in many menus). Give it some time.
It seems to me that you are more scared about resource consumption (e.g. flash) or marginal aspects more than real annoying bugs in all the PRs merged in the last weeks causing freezes everywhere.
To drastically reduce the flash usage it is enough to avoid almost 17 languages in the flash. It's enough to change the code in order to set only one language to import at compilation time

traffic-light, 2020-08-16:
> ### Description
> 
> The 100mm button in tuning menu is not useful as it does not check for heating

See #982
It's more advanced then just extruding. You can immediately calculate new e-steps

rightawy, 2020-08-16:
@digant73  
I specifically linked to the comment and not the actual PR. did you not read that comment? and also, BTT already got rid of all the toggle icons in the settings menu (e.g. language, sound, baud rate). 

> Heat and pheheat menus were drastically changed by the PR author

Is it not a good thing to give an option to set the temperature using the number pad?  at this point, to me, it looks like you are being scared of UI changes. There are many things that were changed drastically and are actually better than their previous implementations.

>and I perfectly agreed with most of the people to not drastically change something that was working fine.

This contradicts your implementation of next button as a toggle.

>It seems to me that you are more scared about resource consumption (e.g. flash) or marginal aspects more than real annoying bugs in all the PRs merged in the last weeks causing freezes everywhere.

Does it mean that if someone points out an issue because he is scared of it? Is this why you are pointing out drastically changing UI?  I request you to please do not bend this into personal opinion thing. 

The list mode was added because a majority of the users wanted it over the icon mode.

>To drastically reduce the flash usage it is enough to avoid almost 17 languages in the flash. It's enough to change the code in order to set only one language to import at compilation time

That is for the program flash. what about the SPI flash which stores the icons?

traffic-light, 2020-08-16:
And I think, just like @digant73 that solving the current bugs is more important than the memory where you're taking about.

(I've got to admit. It's indeed another storage)

traffic-light, 2020-08-16:
A, maybe, good midway to remove the toggle icons could be:
One button with a eeprom logo that opens a menu with all eeprom actions.

But I don't care that much. I like it how it's right now

digant73, 2020-08-16:
@traffic-light that is also my preferred solution to guruatwhal in his current PR. Splitting all the EEPROM features on MachineSettings sacrifice the Terminal menu while it should continue to be there. All the icons in MachineSettings are already other menus, so an EEPROM menu make sense.

NicoletoAerials, 2020-08-17:
I agree, the menu is getting messy. Lost of new features but many have repeated icons. I.e. baby stepping in both the printing main menu and the more menu. Also, can be found in the ABL menu. Z-offset is in the ABL, the baby stepping menu, the machine configuration menu, and so on.

There are other things that cannot be changed via the icons but only the configure.ini file. For example the z height for the manual leveling feature, can only be changed in increments of 0.1 mm in the config.ini file. There is no explanation if such value would have any effect in the whole z offset process or even the ABL process.

PR #1188 (2020-10-28): Mesh Editor + improvements + bugfix (replace PR #1154)
**NEW FEATURES:**
* resolves #288
* resolves #594
* resolves #777
* Implemented #288 and #777 adding a Mesh Editor (under BL menu) and Mesh Tuner menus providing a visual mesh grid (gradient colors are configurable in config.ini) and a fine tune tool for mesh editing. The Mesh Editor supports:
1) MBL
2) UBL
3) ABL Bilinear

**BUG FIXES:**
* Fixed #1167 MBL process causing "Invalid mesh" error trying to load the meshes with M420 S1. MBL process is now based on use of standard move gcode
* Fixed #967 related to wrong Software Endstops enabli [... cut]
closed 2020-10-23 by rightawy