bigtreetech/BIGTREETECH-TouchScreenFirmware #968: [Q] a lot of open issues

traffic-light, 2020-08-13:
There are a lot open/ double issues.

Maybe it's a good idea to close all bugs/ questions that are older than 2 months?

Most of them are not relevant.

Tygrys-1, 2020-08-13:
IMHO this project lacks coordination. Or kind of leadership. Or even the consistent vision of it's future. @bigtreetech ?

traffic-light, 2020-08-13:
In my opinion it's not really something to blame bigtreetech for.
They've got more things to do.

Maybe it's an idea for them to add moderators. Some active users that would like to help them with this 
(If that is possible)

Tygrys-1, 2020-08-13:
Not to blame anyone, it is opensource, it is free to modify for ones purposes, it is overall a good piece of the software, but the freedom of doing anything might become a nightmare ;-)

IMHO a mess in the reported issues is only a part of the wider thing.

traffic-light, 2020-08-13:
I agree with that.

oldman4U, 2020-08-13:
Hi.

I agree 100%

If you check this repository as well as some others (all reps of hardware which i also use) you will find many tickets where i asked the owner to close it because it is outdated, or hardware issue related or just an old issue which is already fixed. In case the owner did not answer, I asked BTT to close it, like this is done by Scott for the Marlin repo. Unfortunately BTT does not react, so i gave up on this.

Without removing unneeded tickets, the ticket system will become unusable very soon. This has nothing to do with Open Source but with the owner of a repository.

traffic-light, 2020-08-17:
It seems to be that @bigtreetech is closing old tickets.
Good job @bigtreetech !

oldman4U, 2020-08-19:
@bigtreetech 

What can i do that you close unneeded tickets also in other repositories?

Down to less than 150 tickets from more than 200 is great but can only be the beginning. The goal should be approximately 20 open tickets per repo - i believe.

closed 2020-09-01 by bigtreetech