bigtreetech/BIGTREETECH-TouchScreenFirmware #978: [Q] Neopixel - LED strip

DrumClock, 2020-08-15:
I connected the LED strip type WS2812B to the Neopixel connector in the TFT 50 panel.
**How can I control this LED strip?**

abacabie, 2020-08-15:
On the TFT35 you can control the LED from touch screen mode, its under "machine" settings, should be the same for the TFT 50.

DrumClock, 2020-08-15:
If the LED strip is connected to the Neopixel connector on the TFT board, then only one LED is expanded based on the "Rotary knob LED" setting. The color settings in the "Machine Settings" menu are not functional for this connector.

abacabie, 2020-08-15:
I see, i have neopixels connected to the skr mini v2 not the TFT screen, someone more knowledgeable than myself should be able to answer this.

DrumClock, 2020-08-15:
**Yes,** if I connect the Neopixel strip to the RUMBA board, everything works as it should according to the settings in Marlin FW.
It would be good to add brightness settings to the FTF menu as well. 
After selecting a color, the brightness is set to maximum (P255).

closed 2020-08-28 by DrumClock