bigtreetech/BIGTREETECH-TouchScreenFirmware #981: Update Marlin with TFT Mode Text

el-quique, 2020-08-15:
Hi. Having connected the TFT 2.4 with the EXP1 and EXP2, I cannot upload Marlin. This is normal?. THX

Board MKS GEN L (Marlin 2.0.4)
TFT BigTreeTech 2.4 v1.1 (FW v24, v25 or v26)
[Marlin_2.0.4.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5079692/Marlin_2.0.4.zip)

bigtreetech, 2020-08-17:
Do you mean you can't upload Marlin to MKS-Gen-L via USB after connected TFT24?
If so, It should be caused by connected the RS232 cable, not EXP-1 and EXP-2.
it's because that the MKS-Gen-L's USB and AUX1(RS232) use the same UART, the USB can't send data to MKS-Gen-L when RS232 connected.
you can goto "Menu->Settings->Connection->Disconnect" interface, the screen will display "Now you can control the printer by your computer", now you can upload Marlin via USB normally

el-quique, 2020-08-17:
Hi. thanks for answering. I am using the disconnect function, but it still does not allow me to upload Marlin x USB to the MKS GEN L. Only if I disconnect the EXP-1 and EXP-2. Marlin is configured with a Full Graphis. Thank you

el-quique, 2020-08-19:
Hi. Solution:
Hold down for 1 or 2second at the start upload the Marlin.
THX

closed 2020-08-19 by el-quique