bigtreetech/BIGTREETECH-TouchScreenFirmware #983: [Q] Using onscreen Layer info

blueeagle69, 2020-08-16:
During a print using TFT mode, the Layer information always remains at 0.00.
Do I need layer information in my Gcode?

I'm using Prusaslicer.

Thanks.

traffic-light, 2020-08-16:
What's your firmware (see settings -> info

blueeagle69, 2020-08-16:
My firmware is dated August 16 2020 (3.026)
TFT35 E3.

Thanks.

traffic-light, 2020-08-16:
Oh wait. You should add  something in slicer. I will post it tomorrow.

blueeagle69, 2020-08-17:
> Oh wait. You should add something in slicer. I will post it tomorrow.

Thank you :)
I added [layer_z] to my Before layer change in Prusaslicer, but haven't printed since that change (if that's it)  😀

Thanks again.

traffic-light, 2020-08-17:
It should be this:
[image]

(i've placed two of them. Some applications I did run before needed one of them.)

blueeagle69, 2020-08-17:
Cheers, I'll put it in later  :)

blueeagle69, 2020-09-15:
Hi.

Just an update to this.
My On-screen layer info is no-longer working.
During the print it remains at 0.00

oldman4U, 2020-09-25:
Hi. A function does not change by itself. What changed and what is the status of the reported issue?

Please let us know

blueeagle69, 2020-09-25:
I wish I knew.
I'm using Prusaslicer with the same Before Layer Change as before when it worked.
I updated the firmware (I think it was a build before the recent one) and the Layer Info stopped working.
But as for my Slicer and gcode, nothing has changed. I am back on the Biqu B1 build now because I'm suffering from the screen switching to TFT mode when starting a print from Marlin. But that's another issue,

oldman4U, 2020-09-27:
It would be important to know the firmware versions and make a new ticket with that information. Do you know how to compile intermediate firmware versions?

blueeagle69, 2020-09-27:
Hi.
I know how to compile from source, if that's what you mean.
Apologies if you meant something else 😃

oldman4U, 2020-09-28:
You wrote that one firmware worked well. So it would be important to find out which one it was and which version does not work. And it would be good to add this information to the new ticket.

blueeagle69, 2020-09-29:
Unfortunately, I can't recall which build it was.
But on saying that I am back using an old build again because of my screen switching to TFT mode when I confirm a print in Marlin, and the Layer info isn't working on that one either when printing from the onboard SD card (mainboard).
So I think, when I reported it as working I must have been printing from an SD card in the TFT.
Because that's the only time it appears to work.

oldman4U, 2020-09-30:
Would it make sense to attach a gcode file which has been working before.

blueeagle69, 2020-10-01:
I not entirely sure.
As previously mentioned, I believe it worked before because I was printing from media connected to the TFT.
It just doesn't work from the mainboard onboard SD.

blueeagle69, 2020-10-01:
Shall I close this open Question ticket as I have an open bug report concerning the Layer Info not working when printing from the mainboard onboard SD.

oldman4U, 2020-10-01:
Feel free to close it whenever you want.

Where did you start the bug report, in Marlin repo?

blueeagle69, 2020-10-02:
Hi.

No here;
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1135#issuecomment-701599394

Thanks 😀

oldman4U, 2020-10-09:
Hi. I believe you can close this ticket now. Thank you

closed 2020-10-10 by blueeagle69