bigtreetech/BIGTREETECH-TouchScreenFirmware #985: [BUG] Issue with Leveling

Guilouz, 2020-08-17:
When I start a leveling on touch screen, it launches 2 leveling in a row.
When I save this 2 leveling after complete, when I restart my printer, nothing is stored.

When I run G29 via Terminal and an M500 all is OK.

It happens since this PR : https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/44088a8cfeac6bdfe482716bd5497adc27400469

I use AUTO_BED_LEVELING_BILINEAR and when I take a closer look at the code in leveling.c :

` while(infoMenu.menu[infoMenu.cur] == menuAutoLeveling)
  {
    key_num = menuKeyGetValue();
    switch(key_num)
    {
      case KEY_ICON_0:
        storeCmd("G28\n");
        if (infoMachineSettings.enableubl == 1)
        {
          storeCmd("G29 P1\n");
          storeCmd("G29 P3\n");
          storeCmd("M118 A1 UBL Complete\n");
        }
        else
        {
          storeCmd("G29\n");
          leveled = true;
        }
        break;`

It seems G29 P1 and G29 P3 are runs however these commands are used when AUTO_BED_LEVELING_LINEAR is used and not for AUTO_BED_LEVELING_BILINEAR

Something else that i noticed, in Leveling menu on touch screen I have 'Unified Bed Leveling' and not 'Bilinear Bed Leveling' if I refer to the translation file.

bigtreetech, 2020-08-17:
Hello, you can set `#define ENABLE_UBL_VALUE` to 0 (disable Unified Bed Leveling) or 2(Auto-detect)
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/221943d1284d876e29677680f1a8752ca1a07b11/TFT/src/User/Configuration.h#L248
Maybe it should be 2 in default, I will test and modify it.
Thanks for you feedback!

Guilouz, 2020-08-17:
@bigtreetech Just try with auto-detect parameters and it's look good.
I make a PR or you do it ?

radek8, 2020-08-17:
I have the same problem.
This option cannot be changed via Config.ini.
You will need to add this option to Config.ini so that it can be set without compiling a new FW

closed 2020-08-19 by bigtreetech