bigtreetech/BIGTREETECH-TouchScreenFirmware #986: control knob LED not working.

PigeonFX, 2020-08-17:
Whenever I try and set the LED colour I get the error:

`Unknown Comand: "M105 R0 U255 B0 P255" `

Using Marlin 2.0.6 and the latest TFT35 Firmware from the BTT Github.

Guilouz, 2020-08-17:
M105 is used to report temperature, see : https://marlinfw.org/docs/gcode/M105.html

Tygrys-1, 2020-08-17:
Use M150 to change LED colors. But, this will set the color of neopixels (or other type) leds connected to the printer itself.
If you get this error using the menu in TFT, that means that Marlin is compiled without M150 support.
To change the color of the knob, please use the features menu. See #870

oldman4U, 2020-08-18:
Hi

Have you been able to change the color of the knob using the correct menu?

PigeonFX, 2020-08-18:
> Hi
> 
> Have you been able to change the color of the knob using the correct menu?

Yep

closed 2020-08-18 by PigeonFX