bigtreetech/BIGTREETECH-TouchScreenFirmware #987: [Q] How do I get long filenames on the print job screen?

blueeagle69, 2020-08-17:
I have long filenames enabled in Marlin and enabled in the TFT firmware.
I correctly see long filenames when selecting a model to print.
However when the print is in progress the filename reverts to the 8.3 format when shown at the top of the screen.
Can this be changed?

Thanks.
I've included my TFT config just in case.
[config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5086747/config.zip)

blueeagle69, 2020-08-27:
I've also noticed that as well as the filename being in 8.3 on the print progress screen it also reverts to 8.3 on the print confirmation screen.
I'm printing from my onboard SD card.
Thanks

bigtreetech, 2020-09-01:
Hello， Optimizing the firmware of touch screen can solve this problem, Will do this.
Thank you for your feedback

blueeagle69, 2020-09-01:
Thanks for the reply.
I look forward to the improvement  :)

closed 2020-09-03 by blueeagle69