bigtreetech/BIGTREETECH-TouchScreenFirmware #988: [FR] More Filament Runout Customization and options including UART filament runout sensor ports

jedisintraining3, 2020-08-18:
The mainboards you offer support two filament runout sensors, however when I connect them to the TFT they only work while connected to the TFT. However, there is only one TFT Fil_DET port. There are multiple unused UART Ports. I am asking for the ability to use normal and BTT smart sensors attatched to the TFT's UART ports. This will allow for multiple filament runouts sensors while using TFT mode.

It would also be interesting to see more settings available involving filament runout execution like an option for filament runout gcode, or retract length, or length to runout for direct drive extruders, or even the option for XY homing or xyz.

jedisintraining3, 2020-08-18:
@bigtreetech

bigtreetech, 2020-08-19:
#984 in doing.
and it will execute Pause command when touch screen detected filament runout, you can set the pause data in here https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/420d631fc43cf12fef1a765f2cf7c6f84c5df772/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme/config.ini#L223 by config.ini file

jedisintraining3, 2020-08-19:
Awesome thanks!

oldman4U, 2020-09-11:
Hi my young Padawan.

Do you still need this ticket or can you close i?  Saw your comments in the other ticket and i believe someone working on a dual extruder system will be able to compile the TFT firmware. If not, he will make his own ticket.

Thank you

jedisintraining3, 2020-09-13:
We already worked to compile the firmware, I just left this open since we are still working on the project, like manuals and such. I guess we can close it as the commit was just pulled.

Thanks

closed 2020-09-13 by jedisintraining3

paulanashat22, 2020-10-04:
i made a simple solving for this issue but i know this is not the best solution after filament run out sensor detect that filament is finished the machine will go to the parking and will push the filament up after that i press the rest button on the screen and go to print from sd card and resume printing