bigtreetech/BIGTREETECH-TouchScreenFirmware #994: Fan speed not updating (again)

Thijsvanzwam, 2020-08-18:
### Description

Fanspeed stays at 0% during printing

Installed te latest firmware, self compiled. 

### Steps to reproduce

start any print

**Expected behavior**
fan speed updates

**Actual behavior**
stays at 0%

### Hardware Variant
TFT3,5 v3.0

btt SKR 1.4
Marlin bugfix latest (18-8-2020)

### Additional Information
[Configuration.h.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5092536/Configuration.h.zip)

bigtreetech, 2020-08-19:
Hello， Can the "+" "-" "Full" "Half" button work as expected? 
This problem only occurs during the printing? The fan speed does not update and always displays 0% although the actual fan is already running.

traffic-light, 2020-08-19:
A lot op people it's working with the latest version installed. So I don't understand why it's not working for you

Aleks-MO, 2020-08-19:
@bigtreetech [https://scontent.fhel3-1.fna.fbcdn.net/v/t15.5256-10/p180x540/118259681_316418833009921_5129268042144817033_n.jpg?_nc_cat=110&_nc_sid=1055be&_nc_ohc=JVagzZMjJmcAX-wliUU&_nc_ht=scontent.fhel3-1.fna&oh=7e7d726f83c4caf7714f405af31092f4&oe=5F6465E3](url)

traffic-light, 2020-08-19:
Mmm strange...

Aleks-MO, 2020-08-19:
@bigtreetech  Посмотрел у себя в сохраненных  сборках, эта ошибка появилась с июне. В сборке от 11 апреля этой ошибки нет.  Кто-то здесь уже исправлял  такую ошибку.

sam01508, 2020-08-19:
I downloaded the firmware files last night, hoping to fix the fan issue. tried to compile it myself and it fails straight away.
It comes up with an “error: cannot find template file to configure framework” I’ve done nothing different, it could be a vscode issue because my previous firmware now says the same error.
 There was a vs code update to the st stm32 platform, would not of thought that would make a difference. The F0 fan still shows zero while printing, - + in the menu the speed responds. 
In addition it now also says 28 issues in the c_cpp_properties.json file.

Aleks-MO, 2020-08-19:
@sam01508  c_cpp_properties.json (compiler path and IntelliSense settings) Смотри путь и длинные имена... 
https://code.visualstudio.com/docs/cpp/config-msvc

sam01508, 2020-08-19:
Thank you for your reply, I've looked into it. I can't seem to work it out.

Aleks-MO, 2020-08-19:
@sam01508  Ок"    Всего скорей он обновляется..

bigtreetech, 2020-08-20:
@Aleks-MO 
Hello, can you download and test this [.bin](https://raw.githubusercontent.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme/BIGTREE_TFT35_V3.0.26.x.bin) ？

Aleks-MO, 2020-08-20:
@bigtreetech  
Здравствуйте. Проверил, все хорошо, вентилятор показывает, на печать проверь позже. Где в коде была ошибка? Покажите пожалуйста. 
Спасибо за отклик.

bigtreetech, 2020-08-20:
@Aleks-MO Hello, This .bin is compiled by the latest version code. Please make sure your code is updated to the latest version

Aleks-MO, 2020-08-20:
@bigtreetech
Спасибо за Ваш быстрый ответ и отличную работу, понял, посмотрю. Если все будет работать без сбоя, может поставьте  ее как официальную  ххх,27   ?

Aleks-MO, 2020-08-20:
@Thijsvanzwam 
Тему можно закрыть, все работает, проверил.

sam01508, 2020-08-20:
For the hell of I tried this bin file on my E3 screen (hope it’s ok!?). Anyway the fan speed updates correctly while printing. 
Anyone able to walk me through the compiler path etc settings I don’t seen to have done it right. Firmware etc I’m ok with but some things are out of my league!

sam01508, 2020-08-20:
The latest bin file for the E3 board has the fan speed updating while printing.

Aleks-MO, 2020-08-20:
@sam01508
Все нормально?

sam01508, 2020-08-20:
Yes and no really, fan 0 shows correctly while printing. But if you go into the pre-heat menu the screen freezes and you have to turn the printer on and off to get it working again. 
I have updated to the latest E3 binary, I seem to have wondered in on this post by mistake!

Aleks-MO, 2020-08-20:
@ sam01508
Завтра попробую  на зависание как у вас. Не пробовал этот режим.

charakaSamawry, 2020-08-20:
> Hello， Can the "+" "-" "Full" "Half" button work as expected?
> This problem only occurs during the printing? The fan speed does not update and always displays 0% although the actual fan is already running.

Problem occurs only during printing, fan updates on the info screen when changed whilst not printing.

Now the odd thing, the fan half button works but fan FULL button does not seem to work.. 
Although upon further inspection (whilst not printing) it sounds to me like it is actually putting the fan to 100% whilst still only displaying 50%. (audible difference when half and full buttons are pressed.

The weird part (well to me who doesnt know the code), after pressing full and hearing the fan speed up, the displayed value stays at 50% and im no longer able to increase the value beyond 50%. Press half again and then i can manually increase the displayed value beyond 50% (which it didnt visually change from when pressing half again, though the fan could be heard spinning down to half/50%; now allowing me to manually increase the displayed value beyond 50).

Hopefully some help and not to much word salad

bigtreetech, 2020-08-21:
@charakaSamawry Hello, Are you using the lastest version of source code self compiled firmware?

The source code after 2020.8.19  https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/52d39f43d9379514a1b65a2ed7bcba6462182b7a should fix the fan problem.