bigtreetech/BIGTREETECH-TouchScreenFirmware #1000: [Q] Bootloader TFT50

Rebel3D-CZ, 2020-08-19:
Hi, I have a TFT50 with a faulty CPU. I replaced the CPU with a new one. Is it possible to get a bootloader for TFT50 with the procedure how to upload it?

radek8, 2020-08-20:
maybe someone who has the same display will help and copy the chip's memory :-)
viz
https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/20

radek8, 2020-08-20:
instructions try this one
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT35_V2%20Bootloader%20fix/BTT-TFT35-V2.0-Bootloader-flashing-guide-CH340-debugger.pdf

Rebel3D-CZ, 2020-08-20:
I believe the manufacturer's support is solid and the bootloader will provide its customers :)

Rebel3D-CZ, 2020-08-27:
Is the support really unable to provide customers with a bootloader? Very sad :(

av71966, 2020-09-11:
> Hi, I have a TFT50 with a faulty CPU. I replaced the CPU with a new one. Is it possible to get a bootloader for TFT50 with the procedure how to upload it?
> I have ST-Link. I will connect to the processor.

Hi, i have simmilar problem and have second working tft50 and want to repair broken one. If you help me with copying eeprom i can send it to You

Rebel3D-CZ, 2020-09-12:
Hi, I have another TFT50 in about 2 days. I'll read the bootloader and save it here. Chinese ST-LINK V2 is enough to read. Then I will try to help or I will proceed here, but now I am 2 days on vacation :)

Rebel3D-CZ, 2020-09-17:
Everything you need here :)

https://1drv.ms/u/s!AvBL3Ps7S-g3ktIRlwZE_Jj3X-WJCQ?e=6MwVQf

radek8, 2020-09-17:
Great.
Don't you want to post it on the github page so that it is available to others who might need it?

Rebel3D-CZ, 2020-09-17:
Unfortunately, I don't have much experience with github. I don't know where it can or cannot be given and what the rules are. If you can, put the data where it belongs.

CZ: Radku, vůbec nevím kam bych to měl či neměl dát, s gitem jsem nikdy moc nedělal, jsem rád že sem umím něco napsat a zeptat se :D Ale klidně to dej jak to myslíš, aspoň příště budu vědět. Dík Dan

radek8, 2020-09-17:
OK

av71966, 2020-09-18:
> Everything you need here :)
> 
> https://1drv.ms/u/s!AvBL3Ps7S-g3ktIRlwZE_Jj3X-WJCQ?e=6MwVQf
Thank U! U are my hero)

closed 2020-10-09 by Rebel3D-CZ