bigtreetech/BIGTREETECH-TouchScreenFirmware #1001: How to set this up on a mks gen l

mgrenier25, 2020-08-19:
I am trying to get this TFT35v3 to work with a mks gen l board

I connected both ribbon cables as well as the 5 pin serial

The screen in stuck on mode selection screen, turning/pushing/holding the encoder seems to be doing nothing, reset button works fine, brings me back to the same mode selection screen.

radek8, 2020-08-19:
If you have the display mounted in a box or frame, try removing the display so that it is free.

mgrenier25, 2020-08-19:
It is mounted from one side on a plexiglass, I don't think it is a heat issue.

radek8, 2020-08-19:
It's not about the heat.
The pressure of the frame on the edge of the display can cause a short circuit in the touch layer

mgrenier25, 2020-08-19:
Well I removed it from the plexiglass panel and it is behaving the same way, stuck on mode selection screen.
Tbh I am not even sure this is plugged in the right way, I could barely find any info on how to connect the tft35 to an mks board.

radek8, 2020-08-19:
Touch mode must work even if the printer is not connected. It will only tell you that the printer is not connected

radek8, 2020-08-19:
Attention MKS has connectors EXP1 and EXP2 turned against BTT !!

radek8, 2020-08-19:
Connect the serial line to RS232 on the display and to AUX1 on the MKS GenL
You connect 5v, GND, D1, D2
You leave RST unconnected

mgrenier25, 2020-08-19:
Do I flip both of them or only top one?

radek8, 2020-08-19:
EXP1 and EXP2

radek8, 2020-08-19:
if touch mode does not connect to the printer, swap pins between D1 and D2 of the Aux-1 connector

radek8, 2020-08-19:
Of course Marlin must be properly configured :-)

mgrenier25, 2020-08-19:
Works like a charm, tyvm

radek8, 2020-09-06:
Close this ticket.

closed 2020-09-06 by mgrenier25