bigtreetech/BIGTREETECH-TouchScreenFirmware #995: "Fade height OFF" with BLTouch

PigeonFX, 2020-08-19:
Since adding a BLTouch when my printer boots up I get an error "Fade height OFF" 

I have tried commenting it out in Marlin 
`//#define ENABLE_LEVELING_FADE_HEIGHT 

But doing this brings up first a "No Printer connected" error that is then replaced with "Bed leveling off" and the printer connecting but all the ABL menu has gone.

What am I missing about setting up a BLTouch here?

latest example Marlin 2.0.6 and latest TFT build from here.

Configuration.h
https://pastebin.com/zaMY8J3m

`

Guilouz, 2020-08-19:
It's not an error message but an info message, I think, I have the same because I have set Z fade to 0 because I use leveling for all of my prints. I only use M420 S1 Z0.

digant73, 2020-08-24:
@PigeonFX it's not a bug. It's simply a notification message. If you set and store a Z fade value different than 0 you won't receive that message.
Please, close the bug if everything is clear now

PigeonFX, 2020-08-24:
> @PigeonFX it's not a bug. It's simply a notification message. If you set and store a Z fade value different than 0 you won't receive that message.
> Please, close the bug if everything is clear now

Where do I set Z fade value?

digant73, 2020-08-24:
from the Z Fade menu you find inside the ABL menu

digant73, 2020-08-26:
@PigeonFX If you solved the problem, can you please close this bug report?

closed 2020-08-26 by PigeonFX