bigtreetech/BIGTREETECH-TouchScreenFirmware #999: [BUG] tft24 v1.1 freezes

redgod, 2020-08-19:
the last compilation does not work in tft24 v1.1, after a few minutes the tft freezes and stops working and I cannot find the error

1. [normal firmware installation]
2. [all options work]

**Expected behavior**

Leaving the TV on for long periods

**Actual behavior**

if i leave the tft alone, without touching it, after a few minutes it freezes completely and stops working

### Hardware Variant

### skr mini e3 1.2 & Marlin 2.0.6

Aleks-MO, 2020-08-19:
Ваша прошивка от куда?
Здесь все работает - https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/releases/tag/Vx.x.26

redgod, 2020-08-19:
downloaded from here, the same happens in 2 tft24 v1.1

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme

snakesrules94, 2020-08-19:
Hi there,
same issue with the latest FW : freeze after 1 minute .

Aleks-MO, 2020-08-19:
@redgod По моей ссылке пробовал? 
У меня работает (SKR v 1.3 + TFT24 ), попробуй обнулит установки запустив файл пустой с названием reset.txt. Если не работает, нужно будет подождать, я ошибок не вижу..((( 
У меня осталась старая и самая первая прошивка.

redgod, 2020-08-19:
the problem arises with the firmware that has the gcode file preview option activated, with the previous ones it works perfectly, they work badly with the compilation of a few days ago

I will test the reset.txt file and the sent file

Aleks-MO, 2020-08-19:
Там нет просмотра файла gcode.

redgod, 2020-08-19:
in the latest version downloaded from the official github if you have the preview of gcode as an image

Aleks-MO, 2020-08-19:
Последняя официальная версия от БТТ это здесь - https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/releases/tag/Vx.x.26
Если вы про это - https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme
то эти файлы создал не БТТ , эти файлы от этого человека - https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commits?author=traffic-light

redgod, 2020-08-19:
the github is the same but the person who compiled different? it's the same page, I don't understand

Aleks-MO, 2020-08-20:
Я тоже не понял зачем это сделали.(

snakesrules94, 2020-08-20:
Hello , i've just tried the last commit and freezes are still there .

Aleks-MO, 2020-08-20:
@snakesrules94  БТТ еще не обновила.

traffic-light, 2020-08-20:
Please try this one:https://github.com/traffic-light/BIGTREETECH-TouchScreenFirmware_Peter/tree/updates_for_fw_binairies/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme

maybe it's fixed in the newest one

snakesrules94, 2020-08-20:
@traffic-light : just try your firmware , same results , TFT is frozen after less than 1 min .

traffic-light, 2020-08-20:
What was you're previous version?

It could be related. Can you compile your own with that flag in platformio? That could be an easy solution

redgod, 2020-08-20:
locks continue :(

https://youtu.be/YybcvYuNjc8

snakesrules94, 2020-08-21:
@traffic-light : version that works ok for me  (v1.1.26.x) is the one from 13 Jul 2020.

I've also compile this morning from the latest sources and adding the build flag as discussed before but no changes , still freezing after few seconds.

traffic-light, 2020-08-21:
mmm. That is strange. I can't test it for you what it could be. 
What you could do is downloading older versions and check until which one it is still working.
That way it is easier to find the problem.

snakesrules94, 2020-08-21:
I only see official release from april , and the current version , how to download all the intermediate versions ?

Aleks-MO, 2020-08-21:
@snakesrules94 
Бери друг, это последняя от июня которая работала нормально. 
Вношу просмотр картинок, все ломается, imho это время прерываний, может у тебя "зорче глаз" и найдешь где ошибка...  Я устал...((
https://yadi.sk/d/fHFDKyJ6XaotRA

snakesrules94, 2020-08-21:
@traffic-light : ok , found how to chekcout older versions , will try somes from July to now .
@Aleks-MO : thanks ! i keep your version in case of .

Aleks-MO, 2020-08-21:
@ snakesrules94
Ок" Сравни с последней, корректность что внесли в новую.  Там еще лежит от марта.

snakesrules94, 2020-08-21:
Ok , here is the situation right now :
`commit 9739f9ef2440f3efc26bb34aeefb18e21e7f1ed2
Author: Msq001 <760675063@qq.com>
Date:   Tue Jul 21 09:12:19 2020 +0800
model preview for cura plug-ins (#844)

This version is OK , no freeze .

`commit f5b4450fd732a466648f7689206f2a5c77dc7d81
Author: Msq001 <760675063@qq.com>
Date:   Wed Jul 22 10:45:59 2020 +0800
fix Baudrate select menu (#856)
baudrate menu

Starting from this one , freezing occurs .
Differences between both versions are in the menu to set baudrate , no others differences in the whole code.

I will check no deeper to se if i can find something or at least , a workaround .

traffic-light, 2020-08-21:
Thank you for looking into it. I will check it out when I'm home

FStarDev, 2020-08-24:
This is c code, it's pretty normal that a small change somewhere can make things crash in other unrelated places because when compiled a tiny change means stuff will end up in different places in memory compared to previous version. If you have a bug somewhere that writes to unallocated memory it can make the process freeze or some other cryptic errors. 

So, don't stare yourself blind on what code was changed in the latest commit, it could be some old code with a bug in it that suddenly becomes noticable.

snakesrules94, 2020-08-24:
@FStarDev : there are more than 70 commits since the software starts to freeze in TFT24. That's why my opinion , is , at a first approach , to find the root cause of the crash when is started to go wrong . then , if something is found , try to apply the correction in the latest commit . I dont imagine taking each commit and test if it's ok or not . This is not to end user to validate the software , non regression tests should be done before releasing a commit to public by BTT . I also recognise that open source is a really good approach , to make things evolve , but in that particular case , you must admit that with more than 70 commits in 1 month , it could be difficult to go back to the root causes .

FStarDev, 2020-08-24:
@snakesrules94 : Absolutely a good way to start, I agree. Just saying that the bug might be in some completely different place and might have been sitting there waiting to "explode" for a long time.

ibrahimcengiz, 2020-08-24:
I have the same problem with SKR 1.4Turbo + tft24 v11

blueeagle69, 2020-08-24:
Just compiled the latest build.
Whenever I go into the Preheat menu, the screen locks up.
TFT35 E3

blueeagle69, 2020-08-25:
I was suffering lockups in the Preheat menu.
I did a fresh compile but forgot to add the new updated config.ini
Once I went back and rectified that, my crashes in the Preheat menu are gone and the new filament preheat options are there too.
So if anyone is getting crashes, make sure you add a new config,ini  😃

redgod, 2020-08-25:
> Estaba sufriendo bloqueos en el menú de precalentamiento.
> Hice una nueva compilación pero olvidé agregar el nuevo config.ini actualizado.
> Una vez que volví y lo rectifiqué, mis fallas en el menú Precalentamiento desaparecieron y las nuevas opciones de precalentamiento del filamento también están ahí.
> Entonces, si alguien se bloquea, asegúrese de agregar una nueva configuración, ini 😃

I already add config.ini and it crashes the same in tft24

blueeagle69, 2020-08-25:
> I already add config.ini and it crashes the same in tft24

Apologies, I forgot to add, I am running a TFT35 E3.

BradNagel, 2020-08-27:
Was moving from the official release to latest to see if I could fix a issue I was already having. Turns out someone already fixed my issue plus made it better. Though I also have lock ups. :-(

snakesrules94, 2020-08-27:
It's still freezing with the latest updated binary , no changes ...

Guga1310, 2020-08-27:
I realized that the custom gcode are not with a complete list either, only a few appear.

BradNagel, 2020-08-27:
As long as I stay off the main page and go into a sub menu I have no lock ups. After 30 min in a sub menu I went back to the start up screen and it locks up in two to three mins. Some other things I noticed in this version was graphical corruption at the bottom on the main page. Its a white randomly dotted line. Does not show up all the time but sometimes. Going to a sub menu and back clears it.

BradNagel, 2020-08-28:
Since my issues were on the status screen I started to comment out code till I was able to reach stable code. Doing this I reached a point that I could see that the freeze seems to be in the Status Screen => toggleTool => drawTemperature() function. Commenting out DrawTemperature breaks the screens refreshing but also keeps it running.

BradNagel, 2020-08-28:
Umm... 🧐

Description
the bk_color value is uncertain when text_mode == GUI_TEXTMODE_TRANS, so it will freeze if bk_color beyond LCD_COLOR_COUNT

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/6046b465dcefc6516372e57f4b94323cc0fc4c04

BradNagel, 2020-08-28:
I put in a pull request with these changes but I ended up closing it. This change did seem to help for me. In the end there is still something else going on. Running the screen for a long period of time still ended with it freezing.

snakesrules94, 2020-08-28:
@BradNagel : i was about to tell it .
I,ve tested the latest commit with you modification and i also encountered some freezing but after a more longer period than before . i also recognised the white line at the bottom of the display from time to time. My fear is that it could freeze during a long print and who knows what will happen ...
Hopefully, we have the Marlin mode to work .

Guga1310, 2020-08-28:
> Since my issues were on the status screen I started to comment out code till I was able to reach stable code. Doing this I reached a point that I could see that the freeze seems to be in the Status Screen => toggleTool => drawTemperature() function. Commenting out DrawTemperature breaks the screens refreshing but also keeps it running.
> 
> EDIT:
> 
> Comment out these two lines in Menu.c and things are stable for me.
> if (liveicon->lines[i].text_mode != GUI_TEXTMODE_TRANS)
> GUI_SetBkColor(lcd_colors[liveicon->lines[i].bk_color]);

Since my issues were on the status screen I started to comment out code till I was able to reach stable code. Doing this I reached a point that I could see that the freeze seems to be in the Status Screen => toggleTool => drawTemperature() function. Commenting out DrawTemperature breaks the screens refreshing but also keeps it running. 

just following this step I didn't have any more crashes

BradNagel, 2020-08-29:
Playing around with another idea. Adding checks to make sure the color is within the array its trying to use. Currently waiting to see if it crashes.

      if (sizeof(lcd_colors) > liveicon->lines[i].fn_color)
        GUI_SetColor(lcd_colors[liveicon->lines[i].fn_color]);

      if (liveicon->lines[i].text_mode != GUI_TEXTMODE_TRANS && sizeof(lcd_colors) > liveicon->lines[i].bk_color)
        GUI_SetBkColor(lcd_colors[liveicon->lines[i].bk_color]);

DelScipio, 2020-09-02:
With that change it's expected to be fixed?

BradNagel, 2020-09-02:
After a 11 hour print and no lock ups then yes I expect it to be fixed.

[image]

redgod, 2020-09-02:
just compile the firmware?

traffic-light, 2020-09-02:
Good job @BradNagel !
It's strange that something so simple can fix a big problem.

Did you already do a pull request on this?

BradNagel, 2020-09-02:
Yes and it's already merged. :-)

redgod, 2020-09-03:
last night I compiled the firmware, installed and sent a 25 minute print, at the moment it seems to work fine, I will follow the test
[image]

DelScipio, 2020-09-03:
I want o report that I took my TFT24 from the storage and tried it. Running for 7 hours and looks like is working now.

My main problem is that sometimes in standby after a cold boot it hangs after a while. Gonna test in standby when I finish the print and report back.

@BradNagel thanks for looking after it!

oldman4U, 2020-09-07:
@redgod 

Could you please let us know if this problem is also fixed for you and close this ticket if so.

Thank you

redgod, 2020-09-07:
Hello
I tried it only for 30 minutes and I had no blockages of any type ... only that when I finished printing, it did not do it correctly when it reached 99% pause the printing and wait to continue printing although the figure is finished

oldman4U, 2020-09-07:
At least for two other users it is fixed, so i hope it will work for you also!

fingers crossed;-)

FStarDev, 2020-09-08:
I tested with a 1 hour print just now and it works for me as well. It doesn't lock up anymore. I had instant lockup before this fix as soon as I started a print. 

I did get a few white lines at the bottom of the screen during the print that went away when I pressed the menu button, not sure what's up with that but it's not breaking anything.

oldman4U, 2020-09-08:
Sounds great. Attach a picture.

Now we need @redgod to confirm and close this ticket.

FStarDev, 2020-09-08:
ok, I need to reproduce it again first.. it disappeared when I touched the screen.

redgod, 2020-09-10:
sorry for the delay
I have no more locks

the only problem is it does not finish the printing at 100%, it only does it at 99% and leaves the printing as paused asking if I want to continue ...

If you like, you can close the thread because I have no more locks with the few tests carried out

oldman4U, 2020-09-11:
Hi. 

Good to hear it works. 

You are the only one who can close the ticket so please close it and start a new one if needed. 

Thank you

closed 2020-09-11 by redgod