bigtreetech/BIGTREETECH-TouchScreenFirmware #1018: Solution for this error?

ivindor, 2020-08-24:
[image]
**Your question or details you need. Please describe.**
A clear and concise question or detail you would like to ask. Like Hardware related or firmware related.

**specify the device variant related to your question.**

**Additional context**
Add any other context or screenshots about the question here.

ivindor, 2020-08-24:
starts printing and after a while the printing stops and this error occurs

ivindor, 2020-08-24:
[image]

oldman4U, 2020-08-24:
Which mainboad?
Which firmware board and tft?
New problem and what changed?
Which slicer software?

ivindor, 2020-08-24:
Skr 1.4 turbo 
Firmware Marlin 2.06 and the tft is last update 9438192
yesterday I changed the firmware of the tft to solve the bug of the f0 fan and now this appears
And the slicer is test whit prusaslicer and superslicer

ivindor, 2020-08-24:
[image]

snakesrules94, 2020-08-24:
the gcode file was tested with the previous FW of the screen ? What was the FW version before updating ?

Tygrys-1, 2020-08-24:
1) Try to use different SD card.
2) Try to print from USB
3) Verify, that shown errorr lines are correct in the file. The begining of the lines looks corrupted.

radek8, 2020-08-24:
Will this error appear even if you print from Marlin mode?

ivindor, 2020-08-24:
No appear in touch mode

oldman4U, 2020-09-03:
Hi.

What is the status of this ticket?

Please let us know

closed 2020-09-26 by bigtreetech